bblanchon/ArduinoJson #1342: serializeJSON with char* input element creates memory leak?

nzxhvr, 2020-08-10:
Noticed that the element in my JSON object was a null value after several writes to it.

I have a variable (clientID) declared early in my code as char* and this variable is used as an input to one of my elements in my JSON object doc.  The serialized object is then sent via MQTT.   I noticed that the element contained the correct data for the first few broadcast cycles but then was reporting as "null" .  When I investigated via checking the memoryUsage() then JSON object size kept increasing with every write to that element. (I actually have several changing parameters that are getting written periodically so i bundle all the doc entries together for clarity)

"Section 4.2.7 Replacing members" of the documentation discusses "a memory leak if the old value has associated
memory, for example, if the old value is a string, an array, or an object."  So I was not sure if this can also happen with a char* type.

```
char* clientID = "Client_1";
.....
const int capacity = JSON_OBJECT_SIZE(5);
StaticJsonDocument<capacity> doc;
.....
doc["node_id"] = clientID ;
serializeJson(doc, output_json);
```

I was able to correct the issue by switching to const char* clientID, but from reading the documentation "Section 4.6 Duplication of strings" a char* will perform a copy which i thought would replace the current entry value and not append.

I am using NodeMCU/ESP8266.

bblanchon, 2020-08-10:
Hi @nzxhvr,

Your understanding is correct, ArduinoJson duplicates `char*`, but not `const char*`.

* it duplicates most strings, in case they get deallocated
* it stores `const char*` by pointer to avoid copying string literals (which cannot be deallocated)

It is the expected behavior.

See also:

* [FAQ: I found a memory leak in the library!](https://arduinojson.org/v6/faq/i-found-a-memory-leak-in-the-library/)

Best regards,
Benoit

nzxhvr, 2020-08-10:
Thanks for the response and the helpful link.  Also thanks for developing the capability of JSON for Arduino.  Very cool.

closed 2020-08-10 by nzxhvr

bblanchon, 2020-08-11:
You're welcome, @nzxhvr.
Good luck with your project!