bblanchon/ArduinoJson #1343: Nesting of JsonObjects from multiple JsonDocuments without deep copy (shallow)

smuellener, 2020-08-10:
Hi

According to the ArduinoJsonEBook the JsonObject has reference semantics. Is it possible to nest a JsonObject from a JsonDocument within a different JsonDocument without copying the entire content?

The following shows my intention:

https://wandbox.org/permlink/oQi9ypThsoGRLYLA

If yes, how can this be done and if no would this be an improvement to consider?

Thanks!

bblanchon, 2020-08-11:
Hi Steven,

ArduinoJson v5 worked this way, but I decided to switch to by-copy only on v6.

I'm not opposed to adding this feature, but I don't want to clutter the library with things that only a few persons use.
As far as I can tell, you're only the second person asking for this feature since the release of v6 (I tried to find the issue, but I can't get my hands on it).

The demo on wandbox is cool, but can you describe your use-case a bit more?
I just want to make sure there isn't an easy workaround, and that others could benefit from it.

Then, I think copying should still be the default behavior, and referencing a document from another should be explicitly asked by the programmer.
So, we need to come up with a syntax.
Maybe somesing like: `doc2["Doc1"] = `&obj;`
This weird syntax suggests that there is a risky business with the address of the object, which is good.

Best regards,
Benoit

smuellener, 2020-08-12:
Hello Benoit

Thanks for your quick reply!

I am thinking about the following use case: A library that wraps an object of unknown size and adds some information around it. By using a shared pointer and allocating the JsonDocument of unknown size where the size is actually known, this document can be added easily.

I guess one could call memoryUsage() or capacity() on the unknown object, allocate more memory and then work with the current by-copy mechanism. But from an efficiency standpoint this seems a bit cumbersome, would you not agree?

I added an example on https://wandbox.org/permlink/GYSJrkMvZI87ptd3

Regarding a possible syntax you were suggesting 

`somesing like: doc2["Doc1"] = &obj;`

How about using the JsonObjectConst? It should clearly not be changed if defined like this and would be similar to the (hopefully) well-known difference between (const char*) and (char*)

smuellener, 2021-01-30:
Hello Benoit

I would like to bring this topic up once again. Also, I have created a better and more understandable example for it:

https://wandbox.org/permlink/eRPzScrGDGO52Csm

As far as I understood, a document's content has to be copied when added to another document. This is not very efficient both in terms of memory and time. Do you still not see a good solution to add this functionality?

Thank you!

bblanchon, 2021-01-31:
Hi Steven,

Thank you for bringing this issue up and for providing this example.
I agree this is a legitimate use case, but not a common one, so I treat this feature with low priority.
Don't worry; as long as I keep this issue opened, you know that I'm keeping an eye on it.
I'll get back to you as soon as I find the time to write a prototype.
BTW, I still think that a weird syntax using `&` is a good idea, and I believe it's the right workaround for #1226.

Best regards,
Benoit

vortigont, 2021-05-11:
+1 for such feature.
My use case is receiving a json with nested object(s) and echo back a subset added into a different JsonDocument.

Something like:
src_json => {"obj_id":"Obejct1","data":[{<DATA>}] };

JsonObject destination => {"other_id":"bla-bla", "foo":"bar", "data":[{<SAME_DATA>}] }

Cheers!

lultimouomo, 2021-09-07:
Hi Benoit,

I am copying subobjects as well (basically extracting an opaque subobject I received via serial and sending it via HTTP post), reference semantics would be a nice way to save memory.
As a side note, to avoid heap fragmentation I am using two `StaticJsonDocument<8192>`, called `rxJsonDoc` and `txJsonDoc` (I clear them before each use and I don't experience leak issues); I tried copying the subobject by doing:
```
txJsonDoc = rxJsonDoc["message"]
```
but I got crashes that appeared to be due to stack overflows.
I avoided the issue by changing that to:
```
txJsonDoc.to<JsonVariant>().set(rxJsonDoc["message"].as<JsonVariant>());
```
because I thought that with the first version I was creating a temporary `StaticJsonDocument<8192>` on the stack; is that correct?

bblanchon, 2021-09-07:
Hi @lultimouomo. 
8KB on the stack is too much for most microcontrollers, so I recommend switching to `DynamicJsonDocument`.
Fixed memory allocation shouldn't increase the heap fragmentation anyway.
If that doesn't do it, **please open a new issue**.

lultimouomo, 2021-09-07:
>  8KB on the stack is too much for most microcontrollers, so I recommend switching to DynamicJsonDocument.

My `StaticJsonDocument<8192>`s are statically allocated, they are not on the stack. I get issues only when doing `txJsonDoc = rxJsonDoc["message"]`. I was just wondering if as I suspect doing sub-object copy in this way creates a temporary `JsonDocument`, be it dynamic or static.

closed 2022-04-27 by bblanchon

bblanchon, 2022-12-26:
This feature was added in [ArduinoJson 6.20.0](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.20.0).

commit 3d6c328a (2022-04-27): Add `JsonVariant::link()` (resolves #1343)