bblanchon/ArduinoJson #1344: Need help on ArduinoJson Parsing

jbagaskara900, 2020-08-11:
Hello! This is my first project repository and my first time using GitHub.

I am currently working on a project that uses Arduino Uno R3 and Ethernet Shield R3 to collect data from a server and de-nest the JSON data to make it send-able into a mySerial function to another device.

My problem is: I really wanted to fill the blank spaces nearby each parameter. I tried to use the parseObject and parseArray but still can't find anything. I kindly need

your advice and help. Thanks!

[firstproject.zip](https://github.com/bblanchon/ArduinoJson/files/5058207/firstproject.zip)

bblanchon, 2020-08-12:
Hi @jbagaskara900,

I believe it's a memory allocation error because:

* the JsonBuffer size is quite large (995 B)
* you call `client.readString()` which copies the entire input in RAM (710 B)
* your microcontroller is quite small (2048 B)

My advice:

1. Switch to ArduinoJson 6.16; it's much more memory efficient
2. Never use dynamic memory allocation on AVR (no `String` and no `DynamicJsonDocument`)
3. Pass the `client` directly to ArduinoJson to avoid the extra copy.
4. Read the documentation, [JsonHttpClient.ino](https://arduinojson.org/v6/example/http-client/) in particular.

BTW, if you need some clarification around memory allocation and strings, I invite you to read "The Missing C++ Course" in [Mastering ArduinoJson](https://arduinojson.org/book/).

Best regards,
Benoit

jbagaskara900, 2020-08-12:
Thank you so much for the reply, Benoit! I see now that the memory is the problem in my case, I will use your advice to fix it.

I have a question regarding the ArduinoJson version. If I am currently using version 5, once I installed version 6, will it overwrite all of the functions of the previous version? Or can I have both versions in my computer?

Thanks again!

Best regards,
Julio

bblanchon, 2020-08-12:
Hi Julio,

The Arduino IDE doesn't allow installing multiple versions of the same library.
In hindsight, I regret I didn't create two libraries (e.g., ArduinoJson5 and ArduinoJson6); it would have saved a lot of trouble, but I guess it's too late now.

If you want to install several versions, I recommend downloading [the single-file version](https://github.com/bblanchon/ArduinoJson/releases/download/v6.16.1/ArduinoJson-v6.16.1.h) and save it with your project files.

Best regards,
Benoit

jbagaskara900, 2020-08-12:
Alright, thanks to Benoit for the reply again! And I have tried your advices and it worked perfectly! Thank you so much for your help and your awesome library! Best of luck!

Best regards,

Julio

bblanchon, 2020-08-12:
You're welcome, Julio.
Good luck with your project!

closed 2020-08-12 by bblanchon