bblanchon/ArduinoJson #1346: Support expanding DynamicJsonDocument

arendst, 2020-08-12:
I know this has been answered 18 month ago but I feel I want to emphasize again that the lack of this feature in AJ6 is still holding back Tasmotas move from AJ5 to AJ6.

Not every JSON message lenght can be known in advance and not every environment has plenty of (fragmented) heap space.

A discussion can be found here https://github.com/arendst/Tasmota/issues/9053#issuecomment-671997140

I just wanted you to know. 

BTW AJ5 still does a good job!

bblanchon, 2020-08-12:
Hi @arendst,

Tasmotas is ESP8266 only, right?
Why don't you use `ESP.getMaxFreeBlockSize()` and [`shrinkToFit()`](https://arduinojson.org/v6/api/basicjsondocument/shrinktofit/)?

Best regards,
Benoit

arendst, 2020-08-12:
Ah. That's a good one for deserialization.

It won't work for serialization though where the amount of serialized data is unknown at the start.

bblanchon, 2020-08-12:
If I were you, I would give it a try.

> Allocating a random number like 1KB or 2KB is not a good compromise and it is a ticking bomb.

I strongly disagree with this logic.
On the contrary, setting a fixed budget offers you a **protection** against unexpected input.

bblanchon, 2020-08-12:
I plan on supporting elastic capacity again, but not in v6 (#1074).

Fixed memory allocation is a core design decision for v6, and many optimizations rely on it.
I don't know when I'll start a v7, because I still have a lot of ideas for v6.

arendst, 2020-08-13:
I'm glad you reconsider expanding the DynamicJsonDocument again.

BTW using ``ESP.getMaxFreeBlockSize()`` isn't foolproof either considering these notes: 
```
  From libraries.rst
  ESP.getMaxFreeBlockSize() returns the largest contiguous free RAM block in
  the heap, useful for checking heap fragmentation.  **NOTE:** Maximum
  ``malloc()``able block will be smaller due to memory manager overheads.

  From HeapMetric.ino
  ESP.getMaxFreeBlockSize() does not indicate the amount of memory that is
  available for use in a single malloc call.  It indicates the size of a
  contiguous block of (raw) memory before the umm_malloc overhead is removed.

  It should also be pointed out that, if you allow for the needed overhead in
  your malloc call, it could still fail in the general case. An IRQ handler
  could have allocated memory between the time you call
  ESP.getMaxFreeBlockSize() and your malloc call, reducing the available
  memory.
```
I'm looking forward for ArduinoJson v7.x

closed 2020-08-13 by arendst

bblanchon, 2020-08-13:
Thank's for the info, I thought `malloc(ESP.getMaxFreeBlockSize())`; I wonder why it doesn't subtract the overhead...

Anyway, thanks again for opening this issue as it gives me feedback on how the library is used in the field.
Please, **don't hesitate to contact me** via the Live Chat or email; I'd like to fully understand why you prefer to keep ArduinoJson v5 rather than using a fixed allocation.