bblanchon/ArduinoJson #1348: Collecting JSON Data from a Server, Then Parsing it with Software Serial

jbagaskara900, 2020-08-13:
Hello, Benoit!

In the follow up of my previous issue, I have successfully extract data from my JSON file and display them in the serial monitor. I thank you for that. Then, I tried to transfer (so to speak) the data from my first Arduino UNO (with the ethernet shield) to another Arduino UNO (as my simulator device). I tried to follow your tutorial in another issue: (https://github.com/bblanchon/ArduinoJson/issues/1219).

The JSON file is same as the previous issue in: https://github.com/bblanchon/ArduinoJson/issues/1344

This is my sender code (Arduino UNO with Ethernet Shield):

````
#include <ArduinoJson.h>    //Enable data parsing JSON
#include <Ethernet.h>       //Enable ethernet shield
#include <SPI.h>            // peripheral comms within the shield
#include <SoftwareSerial.h> //allowing serial comms
 SoftwareSerial mySerial (7,8); //SRx, STx
 
void setup() {
  // Serial Init
  Serial.begin(9600); 
  while (!Serial) continue;
  mySerial.begin(4800);//this is done to avoid overflow in the transmission
}

void loop() {

// Ethernet Init
  byte mac[] = {0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED};
  if (!Ethernet.begin(mac)) {
    Serial.println(F("Failed to configure Ethernet"));
    return;
  }
  delay(1000);

  Serial.println(F("Connecting to Server..."));

  // Connect to HTTP server
  EthernetClient client;
  client.setTimeout(10000);
  if (!client.connect("192.168.1.9", 80)) {
    Serial.println(F("Connection to the Destination Failed!"));
    return;
  }

  Serial.println(F("You are Connected"));

  // Send HTTP request
  client.println(F("GET /server-trafo/api/proses/1 HTTP/1.1")); //the data for harvest
  client.println(F("Host: 192.168.1.9")); //destination: server-trafo
  client.println(F("Connection: close")); //seal the connection
  if (client.println() == 0) {
    Serial.println(F("Request Sending Fail!")); //else...
    return;
  }

  // Check HTTP status
  char status[32] = {0};
  client.readBytesUntil('\r', status, sizeof(status));
  if (strcmp(status, "HTTP/1.1 200 OK") != 0) {
    Serial.print(F("Unexpected response: "));
    Serial.println(status);
    return;
  }
    // Allocate the JSON document
  // Use arduinojson.org/v6/assistant to compute the capacity.
  const size_t capacity = JSON_ARRAY_SIZE(3) + JSON_OBJECT_SIZE(2) + 3*JSON_OBJECT_SIZE(13) + 600;
  DynamicJsonDocument doc(capacity);

serializeJson(doc, mySerial); // sends the data to the mySerial connected device
delay(60000); // wait a minute (literally)... then sends another
  

}
````
And here is the receiver code (Arduino UNO)

````
//Uno Receiver

#include<SoftwareSerial.h>
#include <ArduinoJson.h>
SoftwareSerial mySerial(7,8);  //D2, D1 = SRX, STX

void setup() {
// Serial Init
Serial.begin(9600); //normal baudrate
mySerial.begin(4800); // number is less so it won't overflow
while (!Serial) continue;
}
 
void loop() {
  if (mySerial.available())
  {
const size_t capacity = JSON_ARRAY_SIZE(3) + JSON_OBJECT_SIZE(2) + 3*JSON_OBJECT_SIZE(13) + 600;
DynamicJsonDocument doc(capacity); //parse the capacity

//if something is wrong
DeserializationError error = deserializeJson(doc, mySerial);

if (error == DeserializationError::Ok) 
{
  //extract the harvest
  // Extract values
  Serial.println(F("Product in Queue:"));
  Serial.println(F("ID Trafo: "));
  Serial.println(doc["data"][0]["id_trafo"].as<char*>());
  Serial.println(F("Tipe Trafo: "));
  Serial.println(doc["data"][0]["tipe_produk"].as<char*>());
  Serial.println(F("Degree: "));
  Serial.println(doc["data"][0]["degree"].as< float >(), 6);
  Serial.println(F("Speed: "));
  Serial.println(doc["data"][0]["speed"].as<float>(), 6);
  Serial.println(F("Fill: "));
  Serial.println(doc["data"][0]["fill"].as< float >(), 6);
  Serial.println(F("Diameter Prim: "));
  Serial.println(doc["data"][0]["diameter_prim"].as<float>(), 6);
  Serial.println(F("Panjang Kawat Prim: "));
  Serial.println(doc["data"][0]["length_prim"].as<float>(), 6);
  Serial.println(F("Jml Prim: "));
  Serial.println(doc["data"][0]["n_wire_prim"].as<float>(), 6);
  Serial.println(F("Kawat Sambung Primer: "));
  Serial.println(doc["data"][0]["wire_conn_prim"].as<float>(), 6);
  Serial.println(F("Jml Kumparan Primer: "));
  Serial.println(doc["data"][0]["turn_prim"].as< float >(), 6);
  Serial.println(F("Diameter Sec: "));
  Serial.println(doc["data"][0]["diameter_sec"].as< float >(), 6);
  Serial.println(F("Panjang Kawat Sec: "));
  Serial.println(doc["data"][0]["length_sec"].as< float >(), 6);
  Serial.println(F("Jml Kumparan Sec: "));
  Serial.println(doc["data"][0]["turn_sec"].as< float >(), 6);
  
} // closes if(error== Des...)
else
{
Serial.print(F("deserializeJson() failed! Reason: "));
Serial.println(error.c_str());

//flush all bytes from serial buffer
while (mySerial.available()>0)
  mySerial.read();
} //closes else

} //closes if(mySerial.avail)
} // closes void loop
````

The two devices are connected with SoftwareSerial.h. I hope you can give me some help with this part.

Best Regards,

Julio

jbagaskara900, 2020-08-13:
Oh ya, I forgot to put out the problem... The problem in this setup is when I opened my serial monitor in the receiver Arduino, I received literally blank result... I really hope you could help me with this... Thank you so much.

Best Regards,
Julio

bblanchon, 2020-08-13:
Hi Julio,

Before introducing Ethernet to the mix, you must ensure that the serial communication works.
Please have a look at: [How to do serial communication between two boards?](https://arduinojson.org/v6/how-to/do-serial-communication-between-two-boards/)

Best regards,
Benoit

jbagaskara900, 2020-08-14:
Hi Benoit!

Yep, it does work now. Turns out, there was an issue in the serial wire, one of them was broken. Now, I already added the ethernet to the mix. But the problem now is, the serial monitor displays the form, it receives the parsed data from the JSON, but it does not display the numbers and char as intended.

The serial monitor now displays as follow:
````
Product in Queue:
ID Trafo: 

Tipe Trafo: 

Degree: 
0.000000
Speed: 
0.000000
Fill: 
0.000000
Dia_Prim: 
0.000000
l_Kawat_Prim: 
0.000000
n_Prim: 
0.000000
Sbg_Prim: 
0.000000
n_Kum_Prim: 
0.000000
Dia_Sec: 
0.000000
l_Kawat_Sec: 
0.000000
n_Kum_Sec: 
0.000000
Product in Queue:
ID Trafo: 

Tipe Trafo: 

Degree: 
0.000000
Speed: 
0.000000
Fill: 
0.000000
Dia_Prim: 
0.000000
l_Kawat_Prim: 
0.000000
n_Prim: 
0.000000
Sbg_Prim: 
0.000000
n_Kum_Prim: 
0.000000
Dia_Sec: 
0.000000
l_Kawat_Sec: 
0.000000
n_Kum_Sec: 
0.000000
````

My question is: does this happen because of memory allocation issue? or perhaps there is another thing happening throughout the transmission? Both my sender and my receiver sketches utilized 59% and 31% storage space (19214 and 10168 bytes), and their global memory respectively 37% and 29% (768 and 609 bytes), do these memory allocations interfere with the JSON transmission? If it does, I need to re-modify the code since I also wanted to add liquid crystal and keypad functions to the receiver code as well for display.

I really look forward for your reply

Best regards,
Julio

bblanchon, 2020-08-14:
Hi Julio,

I don't think it's a memory issue because, if that was the case, [`deserializeJson()`](https://arduinojson.org/v6/api/json/deserializejson/) would have returned [`NoMemory`](https://arduinojson.org/v6/api/misc/deserializationerror/#nomemory).

I think the problem comes from the way you access each value; the path is probably wrong.

You can easily check the layout of the received document by calling `serializeJsonPretty(doc, Serial)` on the receiver.

Best regards,
Benoit

jbagaskara900, 2020-08-14:
Hi Benoit,

Thank you for your reply. I added the serializeJsonPretty(doc, Serial); on the receiver code just like you've adviced me,

````
if (mySerial.available())
  {
const size_t capacity = JSON_ARRAY_SIZE(3) + JSON_OBJECT_SIZE(2) + 3*JSON_OBJECT_SIZE(13) + 600;
DynamicJsonDocument doc(capacity); //parse the capacity

//if something is wrong
DeserializationError error = deserializeJson(doc, mySerial);
serializeJsonPretty(doc, Serial);
if (error == DeserializationError::Ok) 
{
  //extract the harvest
  // Extract values
  Serial.println(F("Product in Queue:"));
  Serial.println(F("ID Trafo: "));
  Serial.println(doc["data"][0]["id_trafo"].as<char*>());
  Serial.println(F("Tipe Trafo: "));
  Serial.println(doc["data"][0]["tipe_produk"].as<char*>());
  Serial.println(F("Degree: "));
  Serial.println(doc["data"][0]["degree"].as< float >(), 6);
  Serial.println(F("Speed: "));
  Serial.println(doc["data"][0]["speed"].as<float>(), 6);
  Serial.println(F("Fill: "));
  Serial.println(doc["data"][0]["fill"].as< float >(), 6);
  Serial.println(F("Dia_Prim: "));
  Serial.println(doc["data"][0]["diameter_prim"].as<float>(), 6);
  Serial.println(F("l_Kawat_Prim: "));
  Serial.println(doc["data"][0]["length_prim"].as<float>(), 6);
  Serial.println(F("n_Prim: "));
  Serial.println(doc["data"][0]["n_wire_prim"].as<float>(), 6);
  Serial.println(F("Sbg_Prim: "));
  Serial.println(doc["data"][0]["wire_conn_prim"].as<float>(), 6);
  Serial.println(F("n_Kum_Prim: "));
  Serial.println(doc["data"][0]["turn_prim"].as< float >(), 6);
  Serial.println(F("Dia_Sec: "));
  Serial.println(doc["data"][0]["diameter_sec"].as< float >(), 6);
  Serial.println(F("l_Kawat_Sec: "));
  Serial.println(doc["data"][0]["length_sec"].as< float >(), 6);
  Serial.println(F("n_Kum_Sec: "));
  Serial.println(doc["data"][0]["turn_sec"].as< float >(), 6);
  
  
} // closes if(error== Des...)
else
{
Serial.print(F("deserializeJson() failed! Reason: "));
Serial.println(error.c_str());

//flush all bytes from serial buffer
while (mySerial.available()>0)
  mySerial.read();
} //closes else

} //closes if(mySerial.avail)

} // closes void loop
````

but then in the serial monitor, it turns like this:
````
nullProduct in Queue:
ID Trafo: 

Tipe Trafo: 

Degree: 
0.000000
Speed: 
0.000000
Fill: 
0.000000
Dia_Prim: 
0.000000
l_Kawat_Prim: 
0.000000
n_Prim: 
0.000000
Sbg_Prim: 
0.000000
n_Kum_Prim: 
0.000000
Dia_Sec: 
0.000000
l_Kawat_Sec: 
0.000000
n_Kum_Sec: 
0.000000
````

there is a 'null' in there, and I still have no clue what is happening with my JSON data. It worked on the previous issue, but once I added the SoftwareSerial communication, it's turning like this. Could there be any interference between the ArduinoJson and SoftwareSerial in my case?

Best regards,
Julio

bblanchon, 2020-08-15:
Hi Julio,

Please use [`ReadLoggingStream`](https://github.com/bblanchon/ArduinoStreamUtils#logging-read-operations) to see exactly what is received.
See the instructions at the bottom of [this page](https://arduinojson.org/v6/how-to/do-serial-communication-between-two-boards/)

Best regards,
Benoit

jbagaskara900, 2020-08-16:
Hi Benoit,

Thank you so much for the reply! I will read the reference, and I'll see what I can do!

Best regards,
Julio

bblanchon, 2020-08-18:
You're welcome :-)
Don't forget to close this issue when it's done.

closed 2020-08-24 by jbagaskara900