bblanchon/ArduinoJson #1349: nested serializing

MakeSenseIoT, 2020-08-14:
I am trying to send to serial port the following JSON from Arduino Nano 33 BLE) Don't know how to do double nested
{"imu": {"Accel": {"Ax": -1,"Ay": 2,"Az": 3},"Gyro": {"Gx": 4,"Gy": 5,"Gz": 6},"Mag": {"Mx": 4,"My": 5,"Mz": 6}}}
Can you help?

StaticJsonDocument<512> doc; 
JsonObject imu = doc.createNestedObject("imu");
JsonObject Accel = doc.createNestedObject("imu");  //https://arduinojson.org/v6/api/jsondocument/createnestedobject/ 
JsonObject Gyro = doc.createNestedObject("imu");
JsonObject Mag = doc.createNestedObject("imu");

  Accel["Ax"] = Ax;
  Accel["Ay"] = Ay;
  Accel["Az"] = Az;
  serializeJson(doc, Serial);

gives: 
{"imu":{"Ax":-0.065918,"Ay":0.97583,"Az":-0.000977}}

bblanchon, 2020-08-14:
Hi @MakeSenseIoT,

Please ask [my assistant](https://arduinojson.org/v6/assistant/) ;-)

Best regards,
Benoit

MakeSenseIoT, 2020-08-14:
Thanks a lot. Works like a dream.

closed 2020-08-14 by MakeSenseIoT