bblanchon/ArduinoJson #1351: Update JSON Array

regina2898, 2020-08-14:
Hello. To the point, my problem was to update JSON array that deserialized through Software Serial.
```
//Sensor node
#include <SoftwareSerial.h>
#include <ArduinoJson.h>

#define rxPin 5
#define txPin 4
#define setPin 6

SoftwareSerial HC12(rxPin, txPin);

int index;
int mysessionid = 0;

void setup() {
  Serial.begin(19200);
  pinMode(rxPin, INPUT);
  pinMode(txPin, OUTPUT);
  HC12.begin(9600);
  while (!HC12.available()) {delay(1);}
}

void loop() {
  while (HC12.available() > 0) {
    delay(1500);
    while (HC12.read()=='$') {
      checkData();
    }
  }
}

void checkData() {  
  StaticJsonDocument<400> doc;
  DeserializationError error = deserializeJson(doc, Serial);
    if (error) {
      Serial.println(F("deserializeJson() first failed: "));
      Serial.println(error.c_str());
    }
  int sessionid = doc["sessionid"].as<JsonObject>();
  boolean uplink = doc["uplink"].as<JsonObject>();
  JsonArray idmeter = doc["idmeter"].as<JsonArray>();
  JsonArray standmeter = doc["standmeter"].as<JsonArray>();
  JsonArray state = doc["state"].as<JsonArray>();
  int count = idmeter.size();

  if (sessionid > mysessionid) {
    if (uplink == false) {
      for (int i = 0; i < count; i++) {
        if (doc["idmeter"][i] == 1) {
          indeks = i;
          doc["standmeter"][indeks] = 100;
          doc["state"][indeks] = true;
        }
      }
    }
  }
  delay(500);
  HC12.println("&");
  serializeJson(doc, HC12);
  serializeJson(doc, Serial);
}
```

But it only prints the sessionid value and showing Invalid Input Error. 
What might be wrong? May someone help me?
Thanks before

bblanchon, 2020-08-15:
Hi @regina2898,

What's the value of `error`?

The following line is suspicious, please check:

```c++
boolean uplink = doc["uplink"].as<JsonObject>();
```

Best regards,
Benoit

regina2898, 2020-08-16:
hi @bblanchon 
I've changed my code, this is the receiver-sender code (it should be receive the code from node red and forward it to HC12 via software serial -- then receive the modified JSON data from HC12, and print it over serial):

```
//Program Konsentrator
#include <SoftwareSerial.h>
#include <ArduinoJson.h>

#define rxPin 5
#define txPin 4
#define setPin 6

SoftwareSerial HC12(rxPin, txPin);

void setup() {
  Serial.begin(115200);
  HC12.begin(9600);
  
  if (HC12.isListening()) {
    Serial.println("OK");
  } else {
    Serial.println("NOK");
  }
}

struct __attribute__((packed)) SENSOR_DATA {
  int sessionid;
  int N;
  boolean uplink;
  int idmeter[3];
  int standmeter[3];
  boolean state[3];
} sensorData;

void loop() {
  //Parsing paket data (JSON) yang diterima dari Nodered, sekedar lewat.
  while (Serial.available()) {
    StaticJsonDocument<500> doc;
    DeserializationError error = deserializeJson(doc, Serial);
    if (error) {
      Serial.println(F("deserializeJson() first failed: "));
      Serial.println(error.c_str());
    }
    JsonObject sessionid = doc["sessionid"].as<JsonObject>();
    JsonObject N = doc["N"].as<JsonObject>();
    JsonObject uplink = doc["uplink"].as<JsonObject>();
    JsonArray idmeter = doc["idmeter"].as<JsonArray>();
    JsonArray standmeter = doc["standmeter"].as<JsonArray>();
    JsonArray state = doc["state"].as<JsonArray>();

    sensorData.sessionid = doc["sessionid"];
    sensorData.N = doc["N"];
    sensorData.uplink = doc["uplink"];

    for (int i = 0; i < idmeter.size() ; i++) {
      sensorData.idmeter[i] = doc["idmeter"][i];
      sensorData.standmeter[i] = doc["standmeter"][i];
      sensorData.state[i] = doc["state"][i];
    }

    //send to Sensor Node
    uint8_t bs[sizeof(sensorData)];
    memcpy(bs, &sensorData, sizeof(sensorData));

    HC12.println("$$");         //just for indicator
    HC12.write(sizeof(bs));
    HC12.write(bs, sizeof(bs));
    delay(1000);
  }

  delay(1000);
  while (HC12.available()) {
    //baca dari Sensor Node
    if (HC12.read() == '&') {
      while ( ! HC12.available() ) {
        delay(10);
      }
      if ( HC12.read() == '&') {
        byte len =  HC12.read();
        while (HC12.available() < len) {
          delay(10);
        }
        HC12.readBytes((char*)&sensorData, len);

        /*Serial.print("session id : "); Serial.println(sensorData.sessionid);
          Serial.print("N : "); Serial.println(sensorData.N);
          Serial.print("uplink : "); Serial.println(sensorData.uplink);
          Serial.print("idmeter : ");
          for (int i=0; i<sensorData.N; i++){
           Serial.println(sensorData.idmeter[i]);
          }
          Serial.print("standmeter : ");
          for (int i=0; i<sensorData.N; i++){
           Serial.println(sensorData.standmeter[i]);
          }
          Serial.print("state : ");
          for (int i=0; i<sensorData.N; i++){
           Serial.println(sensorData.state[i]);
          }*/

        //Tampilkan di Node-Red format JSON secara serial
        StaticJsonDocument<400> dok;
        JsonObject sesi = dok.createNestedObject("sessionid");
        dok["sessionid"] = sensorData.sessionid;
        JsonObject n = dok.createNestedObject("N");
        dok["N"] = sensorData.N;
        JsonObject upl = dok.createNestedObject("uplink");
        dok["uplink"] = sensorData.uplink;
        JsonArray idm = dok["idmeter"].to<JsonArray>();
        JsonArray stdm = dok["standmeter"].to<JsonArray>();
        JsonArray stt = dok["state"].to<JsonArray>();
        for (int i = 0; i < 3; i++) {
          idm.add(sensorData.idmeter[i]);
          stdm.add(sensorData.standmeter[i]);
          stt.add(sensorData.state[i]);
        }
        serializeJsonPretty(dok, Serial);   //Serialprint to Node-Red*/
        delay(1000);
      }
    }
  }
}
```

and my HC12 receiver and modifier code :
```
//Program Sensor Node 1
#include <SoftwareSerial.h>

#define rxPin 5
#define txPin 4
#define setPin 6

int sessionidku = 0;
int idmeterku = 1;

SoftwareSerial HC12(rxPin, txPin);

void setup() {
  Serial.begin(115200);
  HC12.begin(4800);
}

struct __attribute__((packed)) SENSOR_DATA {
  int sessionid;
  int N;
  boolean uplink;
  int idmeter[3];
  int standmeter[3];
  boolean state[3];
} sensorData;
//volatile boolean haveReading = false;

void loop() {
  delay(1000);
  while (HC12.available() > 0) {
    if (HC12.read() == '$') {
      while ( ! HC12.available() ) {
        delay(1);
      }
      if ( HC12.read() == '$' ) {
        periksaData();
      }
    }
  }
}

void periksaData() {
  int indeks;
  while (HC12.available() < 1) {
    delay(10);
  }
  byte len =  HC12.read();

  while (HC12.available() < len) {
    delay(10);
  }
  HC12.readBytes((char*)&sensorData, len);

  if (sensorData.sessionid > sessionidku) {
    sensorData.sessionid = sensorData.sessionid;
    if (sensorData.uplink == false) {
      for (int i = 0; i < sensorData.N; i++) {
        if (sensorData.idmeter[i] == idmeterku) {
          indeks = i;
          if (sensorData.state[indeks] == false) {
            sensorData.standmeter[indeks] = 100;
            sensorData.state[indeks] = true;
          }
        }
      }
    }
  }

  uint8_t oke[sizeof(sensorData)];
  memcpy(oke, &sensorData, sizeof(sensorData));

  HC12.print("&&"); // && just an indicator that this line is a received message
  HC12.write(sizeof(oke));
  HC12.write(oke, sizeof(oke));
  delay(1000);
}

```
It keeps giving me whether incomplete input or invalid input. I've tried to change the baud rate and serial timeout, but still doesn't worked. What might be wrong? 
Thanks !!

bblanchon, 2020-08-18:
Hi @regina2898,

I'm sorry but you cannot throw two hundred lines of code and expect us to find the bug for you.
Please respect the time of other people: narrow down the problem and ask a specific question.

Best regards,
Benoit

closed 2020-08-18 by regina2898