bblanchon/ArduinoJson #1352: How to delete key/value with index

Findus76, 2020-08-14:
Many thanks for a fantastic job.

I am looking for some feature to effectively remove key / value pairs.
Have a large JSON string that I need to go through often. In order to minimize overhead time, I want to extract information and use output from it.

This is a parts of the JSON string
`{
	"Type": "SystemFunction",
	"Function list": {
		"Sum Current": {
			"R/W": "R",
			"Position": "EnergyMeter",
			"Head Type": "Sensor",
			"Sub Type": "Current",
			"Unit": "A"
		},
		"Active Power L1": {
			"R/W": "R",
			"Position": "EnergyMeter",
			"Head Type": "Sensor",
			"Sub Type": "Active Power",
			"Unit": "W"
		},
		"Active Power L2": {
			"R/W": "W",
			"Position": "EnergyMeter",
			"Head Type": "Sensor",
			"Sub Type": "Active Power",
			"Unit": "W"
		},
		"Active Power L3": {
			"R/W": "R",
			"Position": "EnergyMeter",
			"Head Type": "Sensor",
			"Sub Type": "Active Power",
			"Unit": "W"
		},
		"Sum Active Power": {
			"R/W": "W",
			"Position": "EnergyMeter",
			"Head Type": "Sensor",
			"Sub Type": "Active Power",
			"Unit": "W"
		}
	}
}`

The orginal JSON string is about 7500 characters long.
I would like to loop through and delete all sub key under "Function list"  and remove key/value  that do not contain value "R / W": "W"
And get output like this
`{
	"Type": "SystemFunction",
	"Function list": {
		"Active Power L2": {
			"R/W": "W",
			"Position": "EnergyMeter",
			"Head Type": "Sensor",
			"Sub Type": "Active Power",
			"Unit": "W"
		},
		"Sum Active Power": {
			"R/W": "W",
			"Position": "EnergyMeter",
			"Head Type": "Sensor",
			"Sub Type": "Active Power",
			"Unit": "W"
		}
	}
}`

That is the best way?
Best way for me would be to use index.
To loop through key / value and save index where they occur.
Instead of saving all the Key, it takes up quite a lot of memory.
For example
JSONdoc.remove ("Function list", 0) or JSONdoc.remove (1, 0) for remove part "Sum Current" and so on.

Or I can set up filer functions, e.g.
`{
   "Function list": {
     {
       "X": {
       "R / W": "W"
       "Y": y
       }
     }
   }
}`

Where X means, it does not matter what this key value is.
And Y means that all other parameters below must be included, not just R / W.

Brg

bblanchon, 2020-08-15:
Hi @Findus76,

[JsonObject::remove()](https://arduinojson.org/v6/api/jsonobject/remove/) doesn't accept an index, but it accepts an iterator (even if was missing in the documentation).

In your case, you can write:

```c++
JsonObject fnList = doc["Function list"];
for (auto it=fnList.begin(); it!=fnList.end(); ++it) {
  if (it->value()["R/W"] != "R") {
     fnList.remove(it);
  }
}
```

See online demo: https://wandbox.org/permlink/j91zaHwfJloKmvs5

BTW, remember that [JsonObject::remove()](https://arduinojson.org/v6/api/jsonobject/remove/) removes the member from the object, but doesn't release the associated memory (this is a limitation of the monotonic allocator), so it can create a memory leak if you reuse the [JsonDocument](https://arduinojson.org/v6/api/jsondocument/).

Best Regards,
Benoit

Best regards,
Benoit

bblanchon, 2020-08-15:
PS: you cannot achieve this kind of filtering with `DeserializationOption::Filter`.

Findus76, 2020-08-16:
Thanks!

I didn't now that it was possible to remove object in same seatch loop.
I will try.
I will not reuse the JsonDocument, thanks for you head up.

ps  would like to see an extension of function in filters.
To make it possible to search for an object regardless of the above key has for name.

Brg

closed 2020-08-16 by Findus76