bblanchon/ArduinoJson #1353: esp-idf v 4.0 build error with 'idf.py'

sagarvegg, 2020-08-14:
I am running esp-idf V.4.0.1
installed arduino as a components, it uses 'ninja'.
Successfully running everything else except one thing,
Arduinojson library.
We are using Arduinojson library V6.

`../components/arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/IteratorReader.hpp:26:56: note:   template argument deduction/substitution failed:
../components/arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/IteratorReader.hpp: In substitution of 'template<class TInput> ArduinoJson6120_110001::IteratorReader<typename TInput::const_iterator> ArduinoJson6120_110001::makeReader(const TInput&) [with TInput = String]':
../components/arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/deserialize.hpp:72:42:   required from 'ArduinoJson6120_110001::DeserializationError ArduinoJson6120_110001::deserialize(ArduinoJson6120_110001::JsonDocument&, TStream&, ArduinoJson6120_110001::NestingLimit) [with TDeserializer = ArduinoJson6120_110001::JsonDeserializer; TStream = String]'
../components/arduino/libraries/ArduinoJson/src/ArduinoJson/Json/JsonDeserializer.hpp:418:39:   required from 'ArduinoJson6120_110001::DeserializationError ArduinoJson6120_110001::deserializeJson(ArduinoJson6120_110001::JsonDocument&, TInput&, ArduinoJson6120_110001::NestingLimit) [with TInput = String]'
../components/arduino/libraries/IOTMasterBoard/src/core.cpp:2635:85:   required from here
../components/arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/IteratorReader.hpp:26:56: error: no type named 'const_iterator' in 'class String'
In file included from ../components/arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/deserialize.hpp:13,
                 from ../components/arduino/libraries/ArduinoJson/src/ArduinoJson/Json/JsonDeserializer.hpp:7,
                 from ../components/arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:33,
                 from ../components/arduino/libraries/ArduinoJson/src/ArduinoJson.h:9,
                 from ../components/arduino/libraries/IOTMasterBoard/src/memory.h:6,
                 from ../components/arduino/libraries/IOTMasterBoard/src/communication.h:10,
                 from ../components/arduino/libraries/IOTMasterBoard/src/tasks.h:4,
                 from ../components/arduino/libraries/IOTMasterBoard/src/core.h:7,
                 from ../components/arduino/libraries/IOTMasterBoard/src/core.cpp:1:
../components/arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/deserialize.hpp: In instantiation of 'ArduinoJson6120_110001::DeserializationError ArduinoJson6120_110001::deserialize(ArduinoJson6120_110001::JsonDocument&, TStream&, ArduinoJson6120_110001::NestingLimit) [with TDeserializer = ArduinoJson6120_110001::JsonDeserializer; TStream = String]':
../components/arduino/libraries/ArduinoJson/src/ArduinoJson/Json/JsonDeserializer.hpp:418:39:   required from 'ArduinoJson6120_110001::DeserializationError ArduinoJson6120_110001::deserializeJson(ArduinoJson6120_110001::JsonDocument&, TInput&, ArduinoJson6120_110001::NestingLimit) [with TInput = String]'
../components/arduino/libraries/IOTMasterBoard/src/core.cpp:2635:85:   required from here
../components/arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/StdStreamReader.hpp:29:24: note: candidate: 'ArduinoJson6120_110001::StdStreamReader ArduinoJson6120_110001::makeReader(std::istream&)'
 inline StdStreamReader makeReader(std::istream& input) {
                        ^~~~~~~~~~
../components/arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/StdStreamReader.hpp:29:24: note:   no known conversion for argument 1 from 'String' to 'std::istream&' {aka 'std::basic_istream<char>&'}
In file included from ../components/arduino/libraries/ArduinoJson/src/ArduinoJson/Array/ArrayIterator.hpp:8,
                 from ../components/arduino/libraries/ArduinoJson/src/ArduinoJson/Array/ArrayRef.hpp:8,
                 from ../components/arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:17,
                 from ../components/arduino/libraries/ArduinoJson/src/ArduinoJson.h:9,
                 from ../components/arduino/libraries/IOTMasterBoard/src/memory.h:6,
                 from ../components/arduino/libraries/IOTMasterBoard/src/communication.h:10,
                 from ../components/arduino/libraries/IOTMasterBoard/src/tasks.h:4,
                 from ../components/arduino/libraries/IOTMasterBoard/src/core.h:7,
                 from ../components/arduino/libraries/IOTMasterBoard/src/core.cpp:1:
../components/arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantRef.hpp: In instantiation of 'typename ArduinoJson6120_110001::enable_if<(((! ArduinoJson6120_110001::is_same<T, ArduinoJson6120_110001::ArrayRef>::value) && (! ArduinoJson6120_110001::is_same<T, ArduinoJson6120_110001::ObjectRef>::value)) && (! ArduinoJson6120_110001::is_same<T, ArduinoJson6120_110001::VariantRef>::value)), typename ArduinoJson6120_110001::VariantAs<T>::type>::type ArduinoJson6120_110001::VariantRef::as() const [with T = String; typename ArduinoJson6120_110001::enable_if<(((! ArduinoJson6120_110001::is_same<T, ArduinoJson6120_110001::ArrayRef>::value) && (! ArduinoJson6120_110001::is_same<T, ArduinoJson6120_110001::ObjectRef>::value)) && (! ArduinoJson6120_110001::is_same<T, ArduinoJson6120_110001::VariantRef>::value)), typename ArduinoJson6120_110001::VariantAs<T>::type>::type = String]':
../components/arduino/libraries/ArduinoJson/src/ArduinoJson/Object/MemberProxy.hpp:70:52:   required from 'typename ArduinoJson6120_110001::VariantAs<T>::type ArduinoJson6120_110001::MemberProxy<TParent, TStringRef>::as() const [with TValue = String; TObject = const ArduinoJson6120_110001::ElementProxy<ArduinoJson6120_110001::JsonDocument&>&; TStringRef = const char*; typename ArduinoJson6120_110001::VariantAs<T>::type = String]'
../components/arduino/libraries/IOTMasterBoard/src/core.cpp:2643:67:   required from here
../components/arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantRef.hpp:240:24: error: no matching function for call to 'variantAs<String>(ArduinoJson6120_110001::VariantData* const&)'
     return variantAs<T>(_data);
            ~~~~~~~~~~~~^~~~~~~
In file included from ../components/arduino/libraries/ArduinoJson/src/ArduinoJson/Operators/VariantOr.hpp:9,
                 from ../components/arduino/libraries/ArduinoJson/src/ArduinoJson/Operators/VariantOperators.hpp:9,
                 from ../components/arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantRef.hpp:12,
                 from ../components/arduino/libraries/ArduinoJson/src/ArduinoJson/Array/ArrayIterator.hpp:8,
                 from ../components/arduino/libraries/ArduinoJson/src/ArduinoJson/Array/ArrayRef.hpp:8,
                 from ../components/arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:17,
                 from ../components/arduino/libraries/ArduinoJson/src/ArduinoJson.h:9,
                 from ../components/arduino/libraries/IOTMasterBoard/src/memory.h:6,
                 from ../components/arduino/libraries/IOTMasterBoard/src/communication.h:10,
                 from ../components/arduino/libraries/IOTMasterBoard/src/tasks.h:4,
                 from ../components/arduino/libraries/IOTMasterBoard/src/core.h:7,
                 from ../components/arduino/libraries/IOTMasterBoard/src/core.cpp:1:
../components/arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantAs.hpp:55:59: note: candidate: 'template<class T> typename ArduinoJson6120_110001::enable_if<ArduinoJson6120_110001::is_integral<T>::value, T>::type ArduinoJson6120_110001::variantAs(const ArduinoJson6120_110001::VariantData*)'
 inline typename enable_if<is_integral<T>::value, T>::type variantAs(
                                                           ^~~~~~~~~
../components/arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantAs.hpp:55:59: note:   template argument deduction/substitution failed:
../components/arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantAs.hpp: In substitution of 'template<class T> typename ArduinoJson6120_110001::enable_if<ArduinoJson6120_110001::is_integral<T>::value, T>::type ArduinoJson6120_110001::variantAs(const ArduinoJson6120_110001::VariantData*) [with T = String]':
../components/arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantRef.hpp:240:24:   required from 'typename ArduinoJson6120_110001::enable_if<(((! ArduinoJson6120_110001::is_same<T, ArduinoJson6120_110001::ArrayRef>::value) && (! ArduinoJson6120_110001:: [... cut ...] 001::VariantAs<T>::type = String]'
../components/arduino/libraries/IOTMasterBoard/src/core.cpp:2643:67:   required from here
../components/arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantAsImpl.hpp:32:65: error: no type named 'type' in 'struct ArduinoJson6120_110001::enable_if<false, String>'
[1145/1178] Building CXX object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/libraries/IOTMasterBoard/src/spp_ota.cpp.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1`

sagarvegg, 2020-08-14:
@bblanchon i think the most probable issue should be compiler, if i try to compile the same thing with 'make' it's working, but if i go with 'idf.py' which is ninja it's creating a problem.
this is not the issue of library you can close it if you want, help me if you know something about it

bblanchon, 2020-08-14:
Hi @sagarvegg,

To use Arduino's `String` class in a non-Arduino environment, you must set [ARDUINOJSON_ENABLE_ARDUINO_STRING](https://arduinojson.org/v6/api/config/enable_arduino_string/) to `1`.

Best regards,
Benoit