bblanchon/ArduinoJson #1355: Fields serialized as "null" when including other library that messes with nullptr definition.

dr-gino, 2020-08-17:
Targeting ESP8266 using PlatformIO.
Long story short: I was including a library (Arduino-Temperature-Control-Library) before including ArduinoJson.h and it caused ArduinoJson to always generate null values for some fields that were supposed to contain non-0 integer values. (At first I though there were memory issues, but there was plenty available).
It seems that Arduino-Temperature-Control-Library does the following in its header file:
```
#ifndef nullptr
#define nullptr NULL
#endif
```

Including ArduinoJson.h before the other library fixes this, and obviously just redefining nullptr like that is bad behavior. However this bug took me hours to resolve as no obvious errors/warnings were thrown.

Two questions:

1. How exactly does it cause the behavior I was seeing?
2. Can anything be done in ArduinoJson to warn the user/be more defensive? The Arduino Temperature Control Library is pretty popular for reading out DS18b20 temperature sensors. I might not be the first one encountering this issue.

bblanchon, 2020-08-18:
Hi @dr-gino,

Indeed `#ifndef nullptr` is a mistake: the condition is always false because `nullptr` is not a macro.
I saw that https://github.com/milesburton/Arduino-Temperature-Control-Library/pull/155 is trying to fix this bug.

I can add a warning and disable support for `nullptr` accordingly, but I don't manage to reproduce the issue.
Can you share a minimal example with wandbox?
Here is a starting point: https://wandbox.org/permlink/W6P7jhJSPQaHscOA

Best regards,
Benoit

dr-gino, 2020-08-19:
I'm unable to reproduce it on wandbox, probably because the toolchains are very different.
Here's some code that reliably reproduces it on an ESP8266 with the ArduinoJson@6.16.1 library.
Both the Arduino IDE and PlatformIO can be used to reproduce the erroneous output.
Unfortunately I don't have the hardware to do on chip debugging so investigating any further is a real pain.

```
#include <Arduino.h>
#define nullptr NULL
#include <ArduinoJson.h>

void setup() {
    Serial.begin(115200);

    StaticJsonDocument<100> jsonDoc;

    jsonDoc.add(1);

    String jsonString;
    serializeJson(jsonDoc, jsonString);

    Serial.println(jsonString);
}

void loop() {
    // do nothing.  
}
```

This prints out: ```[null]``` but if the ```#define nullptr NULL``` is removed it prints ```[1]``` as expected.

bblanchon, 2020-10-19:
This fix was published in [ArduinoJson 6.17.0](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.17.0).