bblanchon/ArduinoJson #1358: Dynamic memory management: silently doing nothing is a real problem in

richnwhit, 2020-08-25:
I'm using this excellent library in an industrial product. It has to work solidly, reliably, 100% of the time, and unfortunately I'm not getting that, due to memory problems.

Working out exactly how much memory is required is very tricky (and would be a maintenance nightmare – I wouldn’t expect someone coming along adding a new field to remember to add it to the memory capacity), and I’m working in a tiny chip so I can’t just give it loads.
So I tend to measure how much it needs in a typical situation using memoryUsage, double it, and use that amount.

But I keep having problems that I can’t reproduce. It works fine in all my unit tests, and in the PC-based library, but when I run it on my target microcontroller, often the JSON will be incomplete, which I suspect is because it’s run out of memory.  Sometimes there are empty objects (“{}”), and sometimes I get very long lines and my parser bombs out after 1000 characters without a linefeed.

Ideal solutions would be:
1. A memory allocator that allocates memory from the heap as it's needed, rather than all at the start.
2. In the existing scheme, the ability to throw an exception or at least print a warning when it runs out of memory rather than silently doing nothing. I can't think why anyone would want it to just silently do nothing!

As a bare minimum, please could someone point out where in the code it checks if it has enough memory, so I could add some code in my project?

bblanchon, 2020-08-25:
Hi @richnwhit,

Thank you for reporting this problem. I'm sorry to read that.

As noted in #1074, I plan on restoring the elastic capacity that we had in v5, but it's going to be in a long time, and it would not help you detect exhausted heap.

You don't specify if the problem arises for serialization or deserialization, but I suspect it's for serialization; otherwise, you can simply check the error status.
You're the second person to ask for a way to detect that the memory pool was too small.
Of course, you could check the return value of `set()` but doing so in every location would be a pain.
Another solution would be to add a boolean on `JsonDocument` that would tell if the memory pool is exhausted.

If you want to throw an exception, you must do so in [MemoryPool](https://github.com/bblanchon/ArduinoJson/blob/6.x/src/ArduinoJson/Memory/MemoryPool.hpp).
However, I will not merge these changes because ArduinoJson doesn't use exception.

@richnwhit, please understand that fixed memory allocation is a core design principle in ArduinoJson 6.
It's annoying for some, but it's hugely beneficial for the majority of users.
Maybe ArduinoJson is not the right library for your project.

Best regards,
Benoit

bblanchon, 2020-09-01:
Hi @richnwhit,

It's been one week since we spoke over the Live Chat.
How things evolved on your side?

Best regards,
Benoit

richnwhit, 2020-09-01:
Benoit,

Thanks for getting back in touch.

I created a custom memory allocator (essentially two lines, using new and delete[]). This solved the problem of it silently doing nothing if it could not allocate the memory in its constructor.  (I was amazed how large a chunk of memory I had to ask for when testing this on Windows, before it would actually fail: 100 GB!).

I could not see a way to reliably check if it ran out of the memory that it did allocate, so instead I added a check at certain places, issuing a warning if it is using 70% or more of its allocated memory. There are a few problems with this: firstly, if it tries and fails to add a large object, it could end up having used less than 70% capacity and I won't see the warning, even though something went wrong; and also it forces me to waste some memory to give it that margin.

Just to clarify my position: I am not against static allocation; I can see why you're doing it, and I'm not even asking for dynamic memory allocation (though it would work better for me, I can see I'm not your primary audience).
What I'm asking for is error handling, so I know if it has gone wrong.

One way you might be able to keep everybody happy might be to make your custom allocator more powerful: if it owned the block of memory, and was also used to obtain the space for each item added, then it could be overloaded in useful ways, such as the dynamic allocation I would like, and maybe in some wacky ways neither of us has considered (e.g. storing metadata in a database).

Hope this is helpful to you. Just to say again, genuinely, it's a great piece of work and thank you so much for making it public.

Richard

_____________ Servomex incoming mail advice _____________
Servomex may, as part of its normal activities, monitor, edit or censor the content of any information and software, transmitted through, or stored on, its facilities. Servomex reserves the right to monitor communications, as required, to protect the legitimate interests of Servomex and its business partners.

UNRESTRICTED DOCUMENT

----Servomex Group Limited, UK Technical and Service Centre, Jarvis Brook, Crowborough, East Sussex, TN6 3FB, England. Company Registered in England: No.2170458 VAT No.: GB 522 6077 63---- EXPORT CONTROL INFORMATION: Resale, onward shipment or diversion of Servomex products to embargoed destinations including but not limited to Iran, Syria, Sudan and North Korea is strictly prohibited. This email has been scanned for all viruses by MessageLabs and contains information from Servomex which may be privileged or confidential. The information is intended to be for the use of the individual(s) or entity named above. If you are not the intended recipient be aware that any disclosure, copying, distribution or use of the contents of this information is prohibited. If you have received this electronic message in error, please notify us immediately. Servomex may, as part of its normal activities, monitor, edit or censor the content of any information and software, transmitted through, or stored on, its facilities.

bblanchon, 2020-09-02:
Hi Richard,

You're asking for finer control of the memory allocation, but it requires opening the API of the memory pool, which is currently impossible.

However, I could add a boolean on `JsonDocument` that tells if the memory pool is exhausted.
Something like `doc.overflowed()` or `doc.exhausted()`.
Would it be a suitable solution for you?

Benoit

richnwhit, 2020-09-02:
Benoit,

Yes, that would do it!

I've also noticed that add() returns a bool that I could check, whereas I've been using '["name"] = value'.  And createNestedArray and createNestedObject return objects that have an isNull() method. If you can reassure me that I could always use the return value of add() and checking isNull() - that this will be reliable even in combination with nested arrays and objects and when I add one document to another - then maybe there's nothing for you to change after all?  I think when I first read the API, I didn't understand the issue about memory, so I used operator[] because it looked neater.

Though checking a final flag would certainly be neater, and less error-prone in maintenance!

Thanks a lot for your support,

Richard

bblanchon, 2020-09-02:
Hi Richard,

Indeed [JsonArray::add()](https://arduinojson.org/v6/api/jsonarray/add/) and [JsonObject::set()](https://arduinojson.org/v6/api/jsonobject/set/), as well as their equivalents in [JsonDocument](https://arduinojson.org/v6/api/jsondocument/) and [JsonVariant](https://arduinojson.org/v6/api/jsonvariant/), all return a boolean that tells if the insertion was successful.
It's part of the documented API, and I don't plan on changing it, so you can safely rely on it.

Best regards,
Benoit

richnwhit, 2020-09-02:
Thanks Benoit.

Just for clarity: Are you expecting to make a change now, to provide an "error happened somewhere" flag as you suggested this morning, or do you think the tests you mentioned in this last email will be enough for me? I just want to know whether to wait for some imminent new release or to go with the current version.

Thanks a lot,

Richard

bblanchon, 2020-09-02:
To clarify: I agree to add the flag **if** you believe it will help you (and presumably other users as well).
However, I'll need a few days to push it to master and several weeks before publishing it as part of a release.
It wasn't clear to me whether you preferred this flag or the return value of `set()` and `add()`.

richnwhit, 2020-09-02:
Benoît,

It would definitely help me to have the new flag, because I can't force some future maintainer to use set() rather than operator[] (but they are very unlikely to remove the new check).

It will also make the code much easier to read - instead of:
   CHECK_JSON_RESULT(doc.set("name", value));

I'll have:
   doc["name"] = value;

as I do now.

I'm happy to use an interim version for a little while (though I'd like to catch up with the formally released version when It comes out).
There is no rush: I'm away for a week starting Friday, so the earliest I could possibly pick this up would be the 14th.

Thanks a lot for your help,

Richard

closed 2020-09-05 by bblanchon

bblanchon, 2020-10-19:
`JsonDocument::overflowed` is now available in [ArduinoJson 6.17.0](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.17.0).

commit 8d379390 (2020-09-05): Added `JsonDocument::overflowed()` (closes #1358)