bblanchon/ArduinoJson #1360: De/Serialize "enum class" type with JSON

kmalisiewicz, 2020-08-28:
the issue is similar to #1256. The only difference is to use "enum class"
`enum class STATUS_ERROR
{
  ERROR_READ = -3,
  ERROR_WRITE = -2,
  ERROR_UNDEFINED = -1,
  UNMOUNTED = 0,
  SUCCESS = 1
}; `

The core issue is implementation of is_enum in is_enum.hpp which is different that std::is_enum (which when used fixes this issue).

bblanchon, 2020-08-28:
Hi @kmalisiewicz,

Unfortunately, `<type_traits>` isn't available in every Arduino core; that's why I wrote a polyfill.
Even if `std::is_enum` was an option, I'm pretty sure most users want to serialize `enum class` as a string and not as an integer.
I think the best way to implement this is through a custom converter, as tracked by #687.

Best regards,
Benoit

kmalisiewicz, 2020-08-31:
Hi @bblanchon 
Thanks for your response. It seems for me that I have to make fork of the repository and make such changes by my own.

Regards,
Krzysztof

bblanchon, 2020-09-01:
Hi Krzysztof,

If I were you, I would simply cast the value until #687 provides a better solution.

Best regards,
Benoit

Bascy, 2020-09-03:
I have done this for specifying a Pinmode in a configuration file as follows:

```
#include "Arduino.h"

enum class PinMode {
  Input = INPUT,
  Output = OUTPUT,
  Input_Pullup = INPUT_PULLUP
};

PinMode from(const char* name) {
  if (strcmp(name, "INPUT") == 0) {
    return PinMode::Input;
  } else if (strcmp(name, "OUTPUT") == 0) {
    return PinMode::Output;
  } else if (strcmp(name, "INPUT_PULLUP") == 0) {
    return PinMode::Input_Pullup;
  } 
  return PinMode::Input;
}
```

And when reading the json file:

`const PinMode pinMode = from(sensor["pinMode"] | "INPUT");`

kmalisiewicz, 2020-09-03:
> If I were you, I would simply cast the value until #687 provides a better solution.

Maybe I'll do this way. 
Thanks

bblanchon, 2020-09-03:
@Bascy, you can simplify like so:

```c++
PinMode from(JsonVariantConst mode) {
  if (mode == "OUTPUT") return PinMode::Output;
  if (mode == "INPUT_PULLUP") return PinMode::Input_Pullup;
  return PinMode::Input;
}
```

And when reading the json file:

```c++
PinMode pinMode = from(sensor["pinMode"]);
```

Bascy, 2020-09-03:
@bblanchon Thanks!

(I was kinda hoping you would have some improvement suggestions when posting this ;) )

kmalisiewicz, 2020-09-03:
> If I were you, I would simply cast the value until #687 provides a better solution.

I came with this and it works as I want to.

    enum class settings {
        value1,
        value2
    };

    settings enumValue;

    template <class T>
    bool readValue(const JsonObject& doc, const char* name, T& value)
    {
        if (doc.containsKey(name)) {
            value = doc[name].as<T>();
            return true;
        }
        return false;
    }

    template <class T>
    bool readEnumValue(const JsonObject& doc, const char* name, T& value)
    {
        uint8_t enumValue = 0;
        if (readValue(doc, name, enumValue)) {
            value = static_cast<T>(fromDoc);
            return true;
        }
        return false;
    }

    readValue(jsonObj, "setting1", intValue);
    readEnumValue(jsonObj, "setting2", enumValue);

closed 2020-09-13 by bblanchon