bblanchon/ArduinoJson #1361: Delete key in object

Findus76, 2020-08-28:
Continued on issue no: # 1352
Your example works, thank you very much!
The task was to delete all objects where the key was " R "
In the objects where key is "W" I want to delete key "R / W" but also "Description"
Even if the key does not contain "R" or "W", I want to interrupt the loop.
I tried to set it to "it = Function_object.end ();"

With the result that an "Exception (28):" and crash are generated.

DeserializationError err = deserializeJson(FunclistDoc, functionList, Listleght);
	if (!err) 
	{
		if(FunclistDoc.size() == 2)
		{
    		const char* Function_Type = FunclistDoc["Type"].as<char*>();
			if (Function_Type)
			{
				if(strcmp(FunclistDoc["Type"].as<char*>(), "SystemFunction") == 0)
				{
					JsonObject Function_object = FunclistDoc["Function list"].as<JsonObject>();
					if (!Function_object.isNull())
					{
						//char* functionName;
						//const char* RW_type;
						//uint16_t cnt = 0;
						uint16_t ListSize;
						ListSize = Function_object.size();
						if(ListSize > 0)
						{
							//uint16_t index[ListSize];
							//uint16_t indexPT = 0;
							bool statusinForLoop = true;
						//	JsonObject Function = FunclistDoc["Function list"];
  							for (auto it=Function_object.begin(); it!=Function_object.end(); ++it) 
							{
								  //Avgör om "R/W" finns och är i rätt typ
  								if (it->value()["R/W"] == "R") 
								{
									Serial.println("Remove object");
  							    	Function_object.remove(it);
								}
  							  	else if(it->value()["R/W"] == "W") 
								{
  							      	//Function_object.remove(it);
									Serial.println("Remove part object");
								//	FunclistDoc.remove("Function list" it->key() "R/W");
								//	FunclistDoc.remove("Function list" it->key() "Description");
								}
								else 
								{
									Serial.printf("R/W can only bee R or W in function %s\r\n", it->key().c_str());
									it = Function_object.end();
									statusinForLoop = false;
								}
  							}
							if(statusinForLoop)
								serializeJsonPretty(FunclistDoc, Serial);

Example of json string before and desired result
`{
	"Type": "SystemFunction",
	"Function list": {
		"Relay ch1": {
			"R/W": "W",
			"Position": "EnergyMeter",
			"Head Type": "Relay",
			"Sub Type": "Main Voltage",
			"State": "ON/OFF",
			"Description": "Solidstate max 2A"
		},
		"Relay ch2": {
			"R/W": "R",
			"Position": "EnergyMeter",
			"Head Type": "Relay",
			"Sub Type": "Main Voltage",
			"State": "ON/OFF",
			"Description": "Solidstate max 2A"
		},
		"Relay ch3": {
			"R/W": "R",
			"Position": "EnergyMeter",
			"Head Type": "Relay",
			"Sub Type": "Main Voltage",
			"State": "ON/OFF",
			"Description": "Solidstate max 2A"
		},
		"Relay ch4": {
			"R/W": "W",
			"Position": "EnergyMeter",
			"Head Type": "Relay",
			"Sub Type": "Main Voltage",
			"State": "ON/OFF",
			"Description": "Solidstate max 2A"
		}
	}
}`

Expect result
`{
	"Type": "SystemFunction",
	"Function list": {
		"Relay ch1": {
			"Position": "EnergyMeter",
			"Head Type": "Relay",
			"Sub Type": "Main Voltage",
			"State": "ON/OFF",
		},
		"Relay ch4": {
			"Position": "EnergyMeter",
			"Head Type": "Relay",
			"Sub Type": "Main Voltage",
			"State": "ON/OFF",
		}
	}
}`

Console when it has crashed
`Checking function file
Remove part object
Remove object
R/W can only bee R or W in function Relay ch3

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x4021d5c4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000000a depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3fff0a90 end: 3fff0d80 offset: 0190
3fff0c20:  3fff3744 00000001 3ffefc48 40211160
3fff0c30:  20373531 3fff348c 3fff348c 3fff3584
3fff0c40:  3fff3764 3fff3754 3fff3744 40216120
3fff0c50:  3fff3282 3ffe9701 3fff3654 3fff0c34
3fff0c60:  3fff33e7 3fff0c70 00000008 3fff33e7
3fff0c70:  3fff3744 3fff0c34 00000000 00000000
3fff0c80:  3fff3282 00000000 00000000 402127bd
3fff0c90:  3fff3734 3fff0c34 3ffefc48 40212a3c
3fff0ca0:  00000310 3ffefc48 3ffefc48 00000273
3fff0cb0:  00000310 3ffefc48 3fff3174 40211338
3fff0cc0:  4021e2e8 00000000 000003e8 00000074
3fff0cd0:  00000000 00000000 00000000 00000000
3fff0ce0:  3fff0e44 4021e2e8 00000000 000003e8
3fff0cf0:  00000074 00000000 00000000 00000000
3fff0d00:  00000000 3fff0e44 3ffe9c7d 402123d8
3fff0d10:  00000000 3ffefc48 3ffe9c7d 402127bd
3fff0d20:  3ffefc48 3ffefba4 3ffef9f0 3fff0dc8
3fff0d30:  3ffefc48 00000001 3ffef9f0 40211df2
3fff0d40:  4021175c feefeffe 4021d8c4 4021d8ac
3fff0d50:  00000000 00000000 feefeffe feefeffe
3fff0d60:  3fffdad0 00000000 3fff0d88 402143ec
3fff0d70:  feefeffe feefeffe 3ffe8540 40100309
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld`

Many thanks!

Brg

bblanchon, 2020-08-29:
Hi @Findus76,

You can use [JsonObject::remove(const char*)](https://arduinojson.org/v6/api/jsonobject/remove/).

Best regards,
Benoit

Findus76, 2020-08-30:
Hi, Benoit

Thanks.

I make it work like this, 
for (JsonObject::iterator it=Function_object.begin(); it!=Function_object.end() && statusinForLoop; ++it) 
							{
								  //Avgör om "R/W" finns och är i rätt typ
  								if (it->value()["R/W"] == "R") 
								{
									Serial.println("Remove object");
  							    	Function_object.remove(it);
								}
  							  	else if(it->value()["R/W"] == "W") 
								{
									Serial.println("Remove part object");
									JsonObject removepart = Function_object[it->key()];
									removepart.remove("R/W");
									removepart.remove("Description");
								}
								else 
								{
									Serial.printf("R/W can only bee R or W in function %s\r\n", it->key().c_str());
									statusinForLoop = false;
								}
  							}
Is that what you think? Or is there a nicer way?

Brg

bblanchon, 2020-08-31:
Hi @Findus76,

This looks correct, but you can simplify the following line:

```c++
JsonObject removepart = Function_object[it->key()];
```

into

```c++
JsonObject removepart = it->value();
```

Best regards,
Benoit

closed 2020-08-31 by Findus76