bblanchon/ArduinoJson #1362: Way to loop through the ArduinoJson file

Sakib1263, 2020-08-28:
Hi Blanchon!

I just wanted to ask a basic question, but I cannot fix it for my code. Here in the **code** provided below, I want to update a Json file in Thingspeak for Bulk-Update sensor data. Now, for the free version, the Thingspeak platform updates data only after every 15 seconds. But I need to collect it every second. So what I do is I feed the JsonBuffer with sensor data per second, so 15 times in 15 seconds. Then the total string is sent to the Thingspeak as a JsonBuffer and the plot is updated. As you can see the code works but I want to implement it using ArduinoJson library v6, but I can't loop through the JsonDocument. The Json format for Thingspeak should be like this:

{
	"write_api_key": "WRITE_API_KEY",
	"updates": [{
			"delta_t": 4,
			"field1": 1.0,
			"field2": "2.0"
		},
		{
			"delta_t": 2,
			"field1": 1.1,
			"field2": 2.2,
			"elevation": 6,
			"status": "Good"
		}
	]
}

Now how can I update the nested array named "updates" with new objects containing sensor data each second (or any other time interval)? The problem I am facing is that the "write_api_key" and "updates" strings should appear once in every Json Document but new objects containing sensor data will be added in each iteration until the JsonDocument is sent to Thingspeak after 15 seconds. But these nested objects containing sensor data are inside the "updates" nested array should only iterate. 

**Code:**
`#include <WiFi.h>
#include <ThingSpeak.h>
#include <Adafruit_BME280.h>

char jsonBuffer[6500] = "["; // Initialize the jsonBuffer to hold data
// Seemingly maximum buffer size for this setup (esp32-TinyPICO in PlatformIO's Arduino Plugin is 6600 Bytes)
// Verify it before reuse

// Network parameters
const char ssid[] = "OQFibre-TMGQHYM";
const char password[] = "50847183";
const char *readAPIKey = "QNE72TM0LR5JCAP1";
const char *writeAPIKey = "D8F0VB5ET8W67A4V";

// ThingSpeak information
char server[] = "api.thingspeak.com"; // ThingSpeak Server
unsigned long channelID = 1116179;

unsigned long lastConnectionTime = 0;              // Track the last connection time
unsigned long lastUpdateTime = 0;                  // Track the last update time
const unsigned long postingInterval = 15L * 1000L; // Post data every 15 seconds (LONG Format)
const unsigned long updateInterval =  1L  * 1000L; // Update once every second (LONG Format)

WiFiClient client; // Initiate the WiFi Client

// Sensor Information
#define SEALEVELPRESSURE_HPA (1013.25)
Adafruit_BME280 bme; // I2C

// Declare Global Variables
float Temperature, Pressure, Humidity, Elevation;

void connectToWiFi()
{
  Serial.println("Connecting to WiFi network: " + String(ssid));
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(1000);
  }

  Serial.println("Connected to the WiFi");
  // Print your device IP address:
  IPAddress ip = WiFi.localIP();
  Serial.print("IP Address: ");
  Serial.println(ip);

  // Print the received signal strength:
  long rssi = WiFi.RSSI();
  Serial.print("Signal strength (RSSI):");
  Serial.print(rssi);
  Serial.println(" dBm");
}

// Updates the ThingSpeakchannel with data
void httpRequest(char *jsonBuffer)
{
  //JSON format for data buffer in the API
  // Format the data buffer as noted above
  char data[6500] = "{\"write_api_key\":\"D8F0VB5ET8W67A4V\",\"updates\":"; // Replace YOUR-CHANNEL-WRITEAPIKEY with your ThingSpeak channel write API key
  strcat(data, jsonBuffer);
  strcat(data, "}");
  // Close any connection before sending a new request
  client.stop();
  String data_length = String(strlen(data) + 1); //Compute the data buffer length
  Serial.println(data);
  // POST data to ThingSpeak
  if (client.connect(server, 80))
  {
    client.println("POST /channels/1116179/bulk_update.json HTTP/1.1"); // Replace YOUR-CHANNEL-ID with your ThingSpeak channel ID
    client.println("Host: api.thingspeak.com");
    client.println("User-Agent: mw.doc.bulk-update (Arduino ESP32)");
    client.println("Connection: close");
    client.println("Content-Type: application/json");
    client.println("Content-Length: " + data_length);
    client.println();
    client.println(data);
  }
  else
  {
    Serial.println("Failure: Failed to connect to ThingSpeak");
  }
  //delay(250); // Wait to receive the response
  client.parseFloat();
  String resp = String(client.parseInt());
  Serial.println("Response code: " + resp); // Print the response code. 202 indicates that the server has accepted the response
  jsonBuffer[0] = '[';                      // Reinitialize the jsonBuffer for next batch of data
  jsonBuffer[1] = '\0';
  lastConnectionTime = millis(); //Update the last conenction time
}

void createJson(char *jsonBuffer)
{
  unsigned long deltaT = (millis() - lastUpdateTime) / 1000;
  strcat(jsonBuffer, "{\"delta_t\":");
  size_t lengthT = String(deltaT).length();
  char temp[6];
  String(deltaT).toCharArray(temp, lengthT + 1);
  strcat(jsonBuffer, temp);
  strcat(jsonBuffer, ",");

  Temperature = bme.readTemperature();
  strcat(jsonBuffer, "\"field1\":");
  lengthT = String(Temperature).length();
  String(Temperature).toCharArray(temp, lengthT + 1);
  strcat(jsonBuffer, temp);
  strcat(jsonBuffer, ",");
  
  Pressure = (bme.readPressure() / 100.0F);
  strcat(jsonBuffer, "\"field2\":");
  lengthT = String(Pressure).length();
  String(Pressure).toCharArray(temp, lengthT + 1);
  strcat(jsonBuffer, temp);
  strcat(jsonBuffer, ",");

  Humidity = bme.readHumidity();
  strcat(jsonBuffer, "\"field3\":");
  lengthT = String(Humidity).length();
  String(Humidity).toCharArray(temp, lengthT + 1);
  strcat(jsonBuffer, temp);
  strcat(jsonBuffer, ",");

  Elevation = bme.readAltitude(SEALEVELPRESSURE_HPA);
  strcat(jsonBuffer, "\"field4\":");
  lengthT = String(Elevation).length();
  String(Elevation).toCharArray(temp, lengthT + 1);
  strcat(jsonBuffer, temp);
  strcat(jsonBuffer, "},");

  // If posting interval time has reached 2 minutes, update the ThingSpeak channel with your data
  if (millis() - lastConnectionTime >= postingInterval)
  {
    size_t len = strlen(jsonBuffer);
    jsonBuffer[len - 1] = ']';
    httpRequest(jsonBuffer);
    Serial.print("Size of the JsonBuffer = ");
    Serial.println(len);
  }

  lastUpdateTime = millis(); // Update the last update time
}

void setup()
{
  // Initialize Serial port
  Serial.begin(2000000);
  Serial.println("Start");

  bme.begin(); // Begin recording the BME280 sensor

  connectToWiFi();          // Call the Function to Initiate WiFi Communication
  ThingSpeak.begin(client); // Initialize ThingSpeak
}

void loop()
{
  // If update time has reached 1 second, then update the jsonBuffer
  if (millis() - lastUpdateTime >= updateInterval)
  {
    createJson(jsonBuffer);
  }
}`

bblanchon, 2020-08-29:
Hi @Sakib1263,

Thank you for taking the time to write such a detailed message.
Unfortunately, I don't understand your question.
Can you summarize it in one sentence?

Thank you for your understanding.

Best regards,
Benoit

Sakib1263, 2020-08-29:
In one sentence, I just want to know how can I **gradually increase the Json file size** by feeding sensor value after each second, then send the whole Json file to a cloud platform after 15 seconds in ArduinoJson v6? The Json file should look like this:

{
"write_api_key": "WRITE_API_KEY",
"updates": [{
"delta_t": 4,
"field1": 1.0,
"field2": "2.0"
},
{
"delta_t": 2,
"field1": 1.1,
"field2": 2.2,
"elevation": 6,
"status": "Good"
}
]
}

bblanchon, 2020-08-29:
[`doc["update"].createNestedObject()`](https://arduinojson.org/v6/api/jsonarray/createnestedobject/) should do the job.

Sakib1263, 2020-08-29:
Thanks for the reply! I still have this confusion: 
"updates" here is not an object. It is a nestedArray and inside this nestedArray, there will be several nestedObjects. I want to add one nestedObject after each iteration to the nestedArray called "updates".

bblanchon, 2020-08-29:
It may be clearer this way:

```c++
JsonArray updates = doc["updates"];
JsonObject newUpdate = updates.createNestedArray();
newUpdate["delta_t"] = 4;
// ...
```

See [JsonArray::createNestedArray()](https://arduinojson.org/v6/api/jsonarray/createnestedobject/)

Sakib1263, 2020-08-29:
Actually I am already doing something like this:

**Code Snippet:**
void createJson()
{
  unsigned long deltaT = (millis() - lastUpdateTime) / 1000;

  doc["write_api_key"] = writeAPIKey;
  JsonArray updates = doc.createNestedArray("updates");
  JsonObject env_params = updates.createNestedObject();
  
  Temperature = bme.readTemperature();
  Pressure = (bme.readPressure() / 100.0F);
  Humidity = bme.readHumidity();
  Elevation = bme.readAltitude(SEALEVELPRESSURE_HPA);

  env_params["delta_t"] = deltaT;
  env_params["field1"] = Temperature;
  env_params["field2"] = Pressure;
  env_params["field3"] = Humidity;
  env_params["field4"] = Elevation;
  ... ... ...

The issue is, the JsonArray called "updates" is not being appended by a new object after each loop. I should get 15 nested objects inside it but I am getting only 1. I think it is because the JsonArray called "updates" should be initialized only once. But it is being initialized in each loop. **How can I initialize it once and then update or append it iteratively?**

**Output after 15 seconds:**
{
  "write_api_key": "D8F0VB5ET8W67A4V",
  "updates": [
    {
      "delta_t": 1,
      "field1": 26.74,
      "field2": 998.182,
      "field3": 48.06641,
      "field4": 126.2134
    }
  ]
}

bblanchon, 2020-08-29:
It looks like the capacity of the [`JsonDocument`](https://arduinojson.org/v6/api/jsondocument/) is too small.
What does the [ArduinoJson Assistant](https://arduinojson.org/v6/assistant/) recommend?

Sakib1263, 2020-08-29:
I have already verified it before using ArduinoJson Assistant. I am using esp32 so, according to ArduinoJson Assistant, the final string size is around 1550 Bytes while I am allocating around 4000 Bytes in the stack for the doc (StaticJsonDocument). I know I am allocating too much but it is just for testing, I will change later.

I think the problem is about initializing and looping as I mentioned in my **previous comment**, not a memory issue. The code is initializing the JsonArray after every loop, so I am only seeing one object inside the array at the end (which was produced during the last iteration). 

I want to **append objects** to the **originally initialized array** during each iteration. Then **after the 15th iteration, I will Serialize the JsonDocument** and send it to the cloud. Then I will **clear the doc and restart the procedure**.

bblanchon, 2020-08-29:
I don't understand...
Are you calling `doc.createNestedArray("updates")` on each iteration?

Sakib1263, 2020-08-29:
Yes. I know I should not do that. I am doing these in each iteration:

`JsonArray updates = doc.createNestedArray("updates");
  JsonObject env_params = updates.createNestedObject();`

Here "updates" is the nestedArray and "env_params" is the nestedObject, as you can see. Now, I want the nestedArray to be declared once and the nestedObject to recur. But the nestedObject is referenced from the nestedArray as the objects are inside the array. If I do not declare the nestedArray inside the loop just before the nestedObject, how else can I reference it?

bblanchon, 2020-08-29:
I'm sorry, I read your message several times but I could not get the sense of it.
Why do you call `doc.createNestedArray("updates")` repeatedly?
You realize that this erases the existing array, right?

Sakib1263, 2020-08-29:
I do realize that and I do not intend to do this. 

The problem is if I declare it at the beginning of the Arduino Code, outside the loop, either of the following 2 things is happening:
1. If I use **doc.clear()** after I send the JsonDocument to the cloud platform, the **link** between the JsonArray "updates" declared beforehand and the JsonDocument is gone and I can **no longer reuse** it to fill objects again.
2. If I **do not clear** the doc, the objects **keep appending** to the array and the JsonDocument's size keeps on increasing until the MCU is out of memory or it crosses the allocated memory limit. 

If there any way I can write this `JsonArray updates = doc.createNestedArray("updates");` outside the loop and then reuse it again and again?

Sakib1263, 2020-08-30:
Sorry, could not solve my issue with your library. Thanks for replying though. I found an alternative.

closed 2020-08-30 by Sakib1263

bblanchon, 2020-08-31:
Thanks for the update, @Sakib1263.
Good luck with your project!