bblanchon/ArduinoJson #1363: How to save json documents in external EEPROM

ferchinas, 2020-08-30:
Hello Benoît:
I would like to save some json files in an external EEPROM that communicates via the I2C bus with ESP32.
What is the most elegant way to do it?
Thank you very much

bblanchon, 2020-08-30:
Hi @ferchinas,

[Custom readers and writers](https://arduinojson.org/news/2019/11/01/version-6-13-0/) are probably the most elegant solution.

Best regards,
Benoit

ferchinas, 2020-08-30:
Hello again! it is definitely a very elegant solution. I'm going to research it and share the results in a few days.
Thank you very much!

bblanchon, 2020-09-13:
Hi @ferchinas,

Did you manage to make it work?

Best regards,
Benoit

closed 2020-09-22 by bblanchon

combs, 2020-12-16:
Hello! I needed to read JSON from an i2c eeprom, so here's the CustomReader I ended up with:

https://gist.github.com/combs/33902fbbbfe360b997c4fc759939d3fa

It uses this library:
https://github.com/sparkfun/SparkFun_External_EEPROM_Arduino_Library

I think you could adapt it to writing the JSON back out fairly easily...