bblanchon/ArduinoJson #1364: Passing JsonArray as a function out parameter in 6.x?

peterpolidoro, 2020-09-02:
Hi. I am curious if there is a way to pass a JsonArray as a function out parameter in version 6.x. I realize return values are preferable to out parameters in functions, but I am trying to get some legacy code to work with 6.x.

In version 5.x, I could use a reference to a pointer to pass a JsonArray as an out parameter. Something like this: 
>char json[] = "[1,2,3]";
StaticJsonBuffer<256> jb;
bool getValue(ArduinoJson::JsonArray * & value)
{
  ArduinoJson::JsonArray & array = jb.parseArray(json);
  value = &array;
}
ArduinoJson::JsonArray * array_ptr;
getValue(array_ptr);

This does not work in ArduinoJson 6 since JsonArray is now a wrapper class instead of a reference.

Is there some other trick I could use to pass a JsonArray through an out parameter or do I need to rewrite the legacy code to use return values instead? Thanks!

bblanchon, 2020-09-03:
Hi,

You must pass the `JsonDocument` by reference:

```c++
char json[] = "[1,2,3]";
StaticJsonDocument<256> doc;

bool getValue(JsonDocument& doc)
{
  deserializeJson(doc, json);
}

getValue(doc);
JsonArray arr = doc.as<JsonArray>();
```

Best regards,
Benoit

peterpolidoro, 2020-09-03:
Hi. Oh I see, thank you. Is it possible to pass only part of a document instead of the entire root document? For example, what about passing an array that is contained within an object or another array? If that is possible, is there any way to pass a reference or a pointer to just the array within the object document through the out parameter without having to make a copy of it?

```c++
char json[] = "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}";
StaticJsonDocument<256> doc;
deserializeJson(doc, json);

bool getDataArrayFromObject(JsonDocument& array_doc)
{
?
}
StaticJsonDocument<256> array_doc;
getDataArrayFromObject(array_doc);
JsonArray arr = array_doc.as<JsonArray>();
 ```

bblanchon, 2020-09-04:
> Is it possible to pass only part of a document instead of the entire root document?

Yes, like this:

```c++
void f(JsonArray array) {

}

f(doc["data"]);
```

> is there any way to pass a reference or a pointer to just the array within the object document through the out parameter without having to make a copy of it?

I don't understand what you mean. Can you rephrase?

peterpolidoro, 2020-09-04:
That passes a reference to an array as an IN parameter to a function, correct? I was wondering if there is a similar way to pass a reference or pointer as an OUT parameter.

Actually just using references to JsonArray seems to work just fine. Sorry for the confusion, the answer was trivial after all.
```c++
char json[] = "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}";
StaticJsonDocument<256> doc;
deserializeJson(doc, json);
void getDataArrayFromObject(JsonArray& array)
{
  array = doc["data"];
}
JsonArray arr;
getDataArrayFromObject(arr);
```

closed 2020-09-04 by peterpolidoro

bblanchon, 2020-09-04:
[JsonArray](https://arduinojson.org/v6/api/jsonarray/) allows read and write access to the array.
In other words, you can modify the array in the function, it will affect the original document.