bblanchon/ArduinoJson #1367: Question: Struct to JSON and to Serial

SuperBoss9, 2020-09-05:
Hi everyone,

I read FAQ and the Doc but it seems I don't understand the flow. I need to send my structure object (_myStr) of my structure (myStr) to Serial as JSON.

Simple usage of serializeJson(_myStr, Serial); returns 'const struct CU::myStr' has no member named 'accept'

It seems that I do something wrong but don't understand what exavtly.

SuperBoss9, 2020-09-05:
Read the book. Now it is clear. I need to add some methods to my structure that will feed the key:value pairs.

closed 2020-09-05 by SuperBoss9

bblanchon, 2020-09-06:
Hi @SuperBoss9,

Thank you very much for purchasing the book.

Indeed, C++ doesn't allow automatic serialization of structs, so you need to do the mapping in your program.

See also:

* [Can I automatically serialize an object? How to convert a struct to a JSON object?](https://arduinojson.org/v6/faq/automatically-serialize-an-object/)
* [How to do serial communication between two boards?](https://arduinojson.org/v6/how-to/do-serial-communication-between-two-boards/)

Best regards,
Benoit