bblanchon/ArduinoJson #1368: Why adding (char*)0 to JsonArray returns false?

dyarkovoy, 2020-09-05:
According to the docs, `JsonVariant::add()` returns
* `true` if the value was successfully added.
* `false` if there was not enough memory in the JsonDocument.

However, when I do `array.add((char*)0)`, the returned value is always `false`, even if the new slot was added successfully. I traced it down to this piece in VariantData.hpp:

```
  template <typename TStoragePolicy>
  bool setString(const char *s, TStoragePolicy storage_policy) {
    if (s) {
      setString(make_not_null(s), storage_policy);
      return true;
    } else {
      setType(VALUE_IS_NULL);
      return false;
    }
  }

```
I guess this assumes that if `s==nullptr`, then we could not allocate mem somewhere earlier, But what if we purposefully pass `nullptr`? 
This may seem like a minor glitch, but my use case is to make sure there's always enough space in the doc when populating it, so I check the return of every `set()` and `add()` and throw if any returns false, to preserve document consistency.

Maybe there's anoter [proper] way to assign `null` to a `JsonVariant`? 

If I'm not overlooking something and this is a glitch indeed, then it raises another question, maybe we should have a special-case-null-variant static, or separate `setNull()` / `addNull()` methods, like the `isNull()` we have already?

bblanchon, 2020-09-05:
Hi @dyarkovoy,

This is not a minor glitch, it's a bug.

By the way, the other way to add a `null` is `add(nullptr)`.

Best regards,
Benoit

dyarkovoy, 2020-09-05:
Hi, @bblanchon, thanks for confirming, 

> By the way, the other way to add a null is add(nullptr).

great, `add(nullptr)` returns `true`. Will use it as a workaround. Thanks.

closed 2020-09-06 by bblanchon

bblanchon, 2020-10-19:
The fix was published in  [ArduinoJson 6.17.0](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.17.0).

commit d04669d0 (2020-09-06): Fixed result of `JsonVariant::set((char*)0)` (fixes #1368)