bblanchon/ArduinoJson #1369: AJ sketch does not compile -- deserializeJson not called correctly?

jimwims, 2020-09-07:
I wrote the below sketch to retrieve certain parameters from OpenWeatherMap every 15 minutes.  If I comment out all the ArduinoJson code (lines 40-53), it compiles and retrieves and prints out the json data from OWM.  When I put the AJ code back in and try to compile it, I get the message "Error compiling for board ESP32S2 Dev Module."  The same happens if I put only the deserialization code (lines 40-46) back in.  So at least one problem is probably in how I am calling deserializeJson in those lines 40-46.  I have spent a large portion of the last two days looking at your video multiple times, and reading both your web site and FAQ and your book, but don't see where I have gone wrong.  Can you tell me?

The board I am using is an Espressif ESP32-S2 Saola WROVER development board.  I am using AJ 6.16.1 in Arduino 1.8.13 on an iMac running Catalina.

Thanks very much.

JTW_WifiClient0_OWM.ino

Compiler output:

In file included from /Users/jim/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/deserialize.hpp:10,
                 from /Users/jim/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Json/JsonDeserializer.hpp:7,
                 from /Users/jim/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:34,
                 from /Users/jim/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:9,
                 from /var/folders/7m/dwr9ltds473_lrq_n__qbw7c0000gp/T/arduino_modified_sketch_562211/JTW_WifiClient0_OWM.ino:1:
/Users/jim/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/Reader.hpp: In instantiation of 'int ArduinoJson6161_11::Reader<TSource, Enable>::read() [with TSource = HTTPClient; Enable = void]':
/Users/jim/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Json/Latch.hpp:38:9:   required from 'void ArduinoJson6161_11::Latch<TReader>::load() [with TReader = ArduinoJson6161_11::Reader<HTTPClient, void>]'
/Users/jim/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Json/Latch.hpp:30:7:   required from 'char ArduinoJson6161_11::Latch<TReader>::current() [with TReader = ArduinoJson6161_11::Reader<HTTPClient, void>]'
/Users/jim/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Json/JsonDeserializer.hpp:47:27:   required from 'char ArduinoJson6161_11::JsonDeserializer<TReader, TStringStorage>::current() [with TReader = ArduinoJson6161_11::Reader<HTTPClient, void>; TStringStorage = ArduinoJson6161_11::StringCopier]'
/Users/jim/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Json/JsonDeserializer.hpp:67:13:   required from 'bool ArduinoJson6161_11::JsonDeserializer<TReader, TStringStorage>::parseVariant(ArduinoJson6161_11::VariantData&, TFilter, ArduinoJson6161_11::NestingLimit) [with TFilter = ArduinoJson6161_11::AllowAllFilter; TReader = ArduinoJson6161_11::Reader<HTTPClient, void>; TStringStorage = ArduinoJson6161_11::StringCopier]'
/Users/jim/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Json/JsonDeserializer.hpp:33:5:   required from 'ArduinoJson6161_11::DeserializationError ArduinoJson6161_11::JsonDeserializer<TReader, TStringStorage>::parse(ArduinoJson6161_11::VariantData&, TFilter, ArduinoJson6161_11::NestingLimit) [with TFilter = ArduinoJson6161_11::AllowAllFilter; TReader = ArduinoJson6161_11::Reader<HTTPClient, void>; TStringStorage = ArduinoJson6161_11::StringCopier]'
/Users/jim/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/deserialize.hpp:65:46:   required from 'ArduinoJson6161_11::DeserializationError ArduinoJson6161_11::deserialize(ArduinoJson6161_11::JsonDocument&, TStream&, ArduinoJson6161_11::NestingLimit, TFilter) [with TDeserializer = ArduinoJson6161_11::JsonDeserializer; TStream = HTTPClient; TFilter = ArduinoJson6161_11::AllowAllFilter]'
/Users/jim/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Json/JsonDeserializer.hpp:676:39:   required from 'ArduinoJson6161_11::DeserializationError ArduinoJson6161_11::deserializeJson(ArduinoJson6161_11::JsonDocument&, TInput&, ArduinoJson6161_11::NestingLimit) [with TInput = HTTPClient]'
/var/folders/7m/dwr9ltds473_lrq_n__qbw7c0000gp/T/arduino_modified_sketch_562211/JTW_WifiClient0_OWM.ino:40:59:   required from here
/Users/jim/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/Reader.hpp:20:21: error: 'class HTTPClient' has no member named 'read'; did you mean 'end'?
     return _source->read();
            ~~~~~~~~~^~~~
            end
Multiple libraries were found for "WiFi.h"
 Used: /Users/jim/Documents/Arduino/hardware/espressif/esp32/libraries/WiFi
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
exit status 1
Error compiling for board ESP32S2 Dev Module.

jimwims, 2020-09-07:
Here's the sketch:

#include <ArduinoJson.h>
#include <WiFi.h>
#include <HTTPClient.h>
 
const char* ssid = "xxx";
const char* password =  "yyy";
 
const String endpoint = "http://api.openweathermap.org/data/2.5/onecall?&lat=40.7&lon=-73.95&units=imperial&exclude=minutely,hourly,daily&APPID=";
const String key = "zzz";

void setup() {
 
  Serial.begin(115200);
 
  WiFi.begin(ssid, password);
 
  while (WiFi.status() != WL_CONNECTED) {
    delay(3000);
    Serial.println("Connecting to WiFi..");
  } 
  Serial.println("Connected to the WiFi network");
}
 
void loop() {
  if ((WiFi.status() == WL_CONNECTED)) { //Check the current connection status
                                         //need to provide for reconnection when not connected
    HTTPClient http;
    http.begin(endpoint + key); //Specify the URL
    int httpCode = http.GET();  //Make the request
    if (httpCode > 0) { //Check for the returning code
        String payload = http.getString();
        Serial.println(httpCode);
        Serial.println(payload);
      }
    else {
      Serial.println("Error on HTTP request");
    }

    StaticJsonDocument<1024> doc;
    DeserializationError error = deserializeJson(doc, http);
    // Test if parsing succeeds.
    if (error) {
      Serial.print(F("deserializeJson() failed: "));
      Serial.println(error.c_str());
      return;
      }
    
    JsonObject current = doc["current"];
    float current_temp = current["temp"];
    JsonObject current_weather_0 = current["weather"][0];
    const char* current_weather_0_description = current_weather_0["description"];
    Serial.println(current_temp);
    Serial.println(current_weather_0_description);
    
    http.end(); //Free the resources
  } 
  delay(900000);  //900000 = 15 minutes
}

bblanchon, 2020-09-07:
Hi @jimwims,

Indeed, you cannot call `deserializeJson(doc, http)` because `HTTPClient` doesn't implement the `Stream` interface.
However, it exposes the underlying `Stream` thought `getStream()`, so you can do:
  
```c++
DeserializationError error = deserializeJson(doc, http.getStream());
```

Unfortunately, this stream doesn't handle [Chunked transfer encoding](https://en.wikipedia.org/wiki/Chunked_transfer_encoding), so make sure you use HTTP/1.0.

You can find a detailed explanation on pages 99-102 of the book (these pages are available in the [Deserialization Tutorial](https://arduinojson.org/v6/doc/deserialization/)).

Best regards,
Benoit

bblanchon, 2020-09-07:
See also: [How to use ArduinoJson with HTTPClient?](https://arduinojson.org/v6/how-to/use-arduinojson-with-httpclient/)

jimwims, 2020-09-08:
Thanks very much for your help!

bblanchon, 2020-09-08:
You're welcome, @jimwims.
Thank you for using ArduinoJson.
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink:

closed 2020-09-08 by bblanchon