bblanchon/ArduinoJson #1370: Getting deserializeJson() failed: NoMemory

donskytech, 2020-09-07:
I am receiving a String Payload like this

`{"id":8,"name":"Nicholas Runolfsdottir V","username":"Maxime_Nienow","email":"Sherwood@rosamond.me","address":{"street":"Ellsworth Summit","suite":"Suite 729","city":"Aliyaview","zipcode":"45169","geo":{"lat":"-14.3990","lng":"-120.7677"}},"phone":"586.493.6943 x140","website":"jacynthe.com","company":{"name":"Abernathy Group","catchPhrase":"Implemented secondary concept","bs":"e-enable extensible e-tailers"}}`

And I used the assistant to compute the object size

```
#include <ArduinoJson.h>

void parseJSON(String sampleString){
  const size_t capacity = JSON_OBJECT_SIZE(2) + JSON_OBJECT_SIZE(3) + JSON_OBJECT_SIZE(5) + JSON_OBJECT_SIZE(8) + 360;
  DynamicJsonDocument doc(capacity);
    // Parse JSON object
  DeserializationError error = deserializeJson(doc, resultString.c_str());
  if (error) {
    Serial.print(F("deserializeJson() failed: "));
    Serial.println(error.c_str());
    return;
  }
}
```

I am getting deserializeJson() failed: NoMemory

bblanchon, 2020-09-08:
Hi @donskytech,

The allocation of the [DynamicJsonDocument](https://arduinojson.org/v6/api/dynamicjsondocument/)'s memory pool failed because you're running out of heap memory.

See:

* [How to reduce memory usage?](https://arduinojson.org/v6/how-to/reduce-memory-usage/)

Best regards,
Benoit

donskytech, 2020-09-09:
Thanks for the link!

closed 2020-09-09 by donskytech