bblanchon/ArduinoJson #1373: ArduinoJson somehow deserializes an invalid input

ValeriiKim, 2020-09-08:
Hi! At first thank you for this great library!
During my project testing I send to my controller via USART an invalid string (for testing purposes):

`{"op": "setPublication", "modID": 1, "varID": 1, "frequency": 200}123dfgdsgsdjgkfo`

whereas my original JSON string should look like this:

`{"op": "setPublication", "modID": 1, "varID": 1, "frequency": 200}`

All this actions happens in this piece of code (see below). `Serial_buffer` is a char array which holds my input string after all activities.
I check that my string looks like input by line `Logger::info(serial_buffer)` which sends this string back to PC.
```cpp
temp_serial_msg = buffer_of_commands.front();
buffer_of_commands.pop_front();
memcpy(serial_buffer, temp_serial_msg.buffer, MAX_LEN_OF_SERIAL_COMMAND);
Logger::info(serial_buffer);
parse_error = deserializeJson(doc, serial_buffer);
if (parse_error == DeserializationError::Ok)
{
    // somehow my program ends up here 
}
```
I am using the stm32f103, and Arduino for its serial monitor.
Thank you in advance.

bblanchon, 2020-09-09:
Hi @ValeriiKim,

That's a peculiarity of ArduinoJson's parser: it stops reading as soon as it reaches the end of the document.

This feature allows for deserializing [JSON streams](https://arduinojson.org/v6/api/misc/serialized/).
It's not something you can do with a `char*`, but the parser behaves the same way for all types of input.

Unless this is a real problem, I plan on keeping this way.

Best regards,
Benoit

ValeriiKim, 2020-09-09:
That's ok, thank you for the answer! I thought that it was problem with my code, but now it's all clear. Apparently I should check unrelated characters by myself.

closed 2020-09-13 by bblanchon