bblanchon/ArduinoJson #1375: Question:  Wrap existing JSON Document

lbussy, 2020-09-08:
Hi again!

Searching for a good way to do this.  I have a JSON doc that I put together programmatically.  It's about 3k or so.  I have a block of code that puts it all together that I would like to re-use with a slightly different outcome.  MailerCode(TM) follows:

```
function createReport(bool doSerial = false)
{
    addJsonStuff here;
    String json;
    serializeJson(doc, json); // {"json":"stuff}
    sendHttpReport(json);

    if (sendSerial)
    {
        wrappedJson = wrapJson(json); // {"wrapper":{"json":"stuff}}
        sendSerialReport(wrappedJson);
    }
}
```
If it was a few hundred chars I'd just do it twice.  The way I understand `JsonDocument::createNestedObject()` I'd be duplicating my space.  Is there a good/better way to do this?

bblanchon, 2020-09-09:
Hi Lee,

You must create a second document that includes the first.

The default way to do that is to duplicate the original document in the enclosing one:

```c++
wrappedJson["wrapped"] = doc;
```

In your case, however, we can avoid the copy because we have a pointer to the serialized string:

```c++
wrappedJson["wrapped"] = serialized(json.c_str());
```

[serialized()](https://arduinojson.org/v6/api/misc/serialized/) obeys to the usual string duplication rules: since we pass a `const char*`, it will not make a copy.

Best regards,
Benoit

lbussy, 2020-09-09:
Thanks Benoit!  Seems so simple now - hindsight being what it is.  I appreciate the assist.

lbussy, 2020-09-09:
Okay maybe I'm being dense, this comes back as null for me:

```
        if (sendSerial)
        {
            DynamicJsonDocument wrapped(JSON_OBJECT_SIZE(1) + capacity);
            JsonObject r = doc.createNestedObject("r");
            r["r"] = serialized(json.c_str());
            serializeJson(r, Serial);
        }
```

It's entirely possible I need to do this after coffee. :)

bblanchon, 2020-09-10:
```c++
DynamicJsonDocument wrapped(JSON_OBJECT_SIZE(1));
wrapped["r"] = serialized(json.c_str());
serializeJson(wrapped, Serial);
```

See demo: https://wandbox.org/permlink/LRbibw8hRxbwjqGD

lbussy, 2020-09-10:
That ... yeah ... sometimes you need to be punched in the face with an answer.  Thank you again sir.

closed 2020-09-10 by lbussy