bblanchon/ArduinoJson #1377: Deserializing from MQTT - deserializeJson() failed: invalid input

ianjmcgregor, 2020-09-11:
Hi Hoping someone can point out where I'm going wrong here

I've been trying to deserialize a JSON string sent from NodeRed via MQTT to an ESP8266 (Wemos D1 Mini)
and I keep getting a deserializeJson() failed: InvalidInput

Using: 
Arduino IDE 1.8.13
ArduinoJson version 6.15.2
pubsubclient v 2.7.0

I'm assuming I've got the JSON string wrong but can't figure out where
The JSON string looks like this

from node red debug (i.e. string being sent)
{"ColHighSP":"6","ManHeat":"4","PotSP":"8"}

from mqtt serial function (function I use when I don't have wemos connected to serial) 
{"ColHighSP":"6","ManHeat":"4","PotSP":"8"}

from serial port
{"ColHighSP":"6","ManHeat":"4","PotSP":"8"}
deserializeJson() failed: InvalidInput

I have checked for unprintable characters in the string by printing the received bytes as integers to the serial port and everything looked fine.

my MQTT callback function where I do the deserialization

`
// Compiled for Wemos D1 mini
// Usually programmed via OTA, but serial can be plugged in if equipment covers are taken off
// Mostly Boiler plate stuff from examples

////////////////////////////////////////////////////////////////////////////////////////////////////
// Handle incoming commands from MQTT
void mqtt_callback(char* topic, byte* payload, unsigned int payloadLength) {

  Serial.print("Message arrived [");
  Serial.print(topic);
  Serial.print("] ");
  Serial.println(payloadLength);

// Check it is the correct Topic  
  if (strcmp(topic, "State") == 0) {

    String payload_buff = "";

    for (int i = 0; i < payloadLength; i++) {
      payload_buff = payload_buff + String((char)payload[i]);
      Serial.println("");
      Serial.print((int)payload[i]); // printed bytes as integers to serial to check for unprintable characters
      Serial.println("");
    }
    payload_buff = payload_buff + '/0';

    // mqttSerialData & mqttNodeSerial are a Variable & a Function to send Debug data to MQTT topic (\<nodename>\info\serial) 
    // when I don't have serial connection
    mqttSerialData = "Message arrived: " + payload_buff;
    mqttNodeSerial();

    // JSON Stuff
    StaticJsonDocument<256> doc;

    DeserializationError error = deserializeJson(doc, payload, payloadLength);

    if (error) {
      Serial.print(F("deserializeJson() failed: "));
      Serial.println(error.c_str());

      return;
    }

    // Fetch values.
    // Most of the time, you can rely on the implicit casts.
    // In other case, you can do doc["time"].as<long>();

    int HeatSet_JSON = doc["ManHeat"];
    float ColHighSPJSON = doc["ColHighSP"];
    float PotSP = doc["PotSP"];

    // mqttSerialData & mqttNodeSerial are a Variable & a Function to send Debug data to MQTT topic (\<nodename>\info\serial) 
    // when I don't have serial connection
    mqttSerialData = "JSON Data Arrived";
    mqttNodeSerial();
    mqttSerialData = String(HeatSet_JSON).c_str();
    mqttNodeSerial();
    mqttSerialData = String(ColHighSPJSON).c_str();
    mqttNodeSerial();
    mqttSerialData = String(PotSP).c_str();
    mqttNodeSerial();

  }
}
`

My thanks for any help in advance

Ian

bblanchon, 2020-09-11:
Hi,

I'm sorry, I cannot spot the problem.

There are many irrelevant stuffs in this snippet, like the weird gymnastic to print the payload.
Can you reduce the snippet to the bare minimum?
Also, ~~double-check~~ triple-check the content of `payload`.

Best regards,
Benoit

closed 2020-09-19 by bblanchon

ianjmcgregor, 2020-09-23:
Used same MQTT topic from Node-Red, in another ESP8266 pretty much the same code above, just leaving out the application specific code (yes I still had the weird gymnastic to print the payload - that came from another example sketch in my Arduino libraries), and leaving just the diagnostic serial output and it worked?

I'm assuming I have some minor typo somewhere in the original Sketch which I will dig into

Thanks Benoit, for having a look

bblanchon, 2020-09-23:
You're welcome.
Thanks for the update, Ian.