bblanchon/ArduinoJson #1379: DeserializationError throws IncompleteInput from config.json

unbreakmat, 2020-09-14:
Hi 

I'm using arduinojson v6.16.1
I've made a file in the data folder naming config.json
It contains this: 
`{"mqtt_host":"10.0.254.100"}`

i uploaded this file to my NodeMCU 1.0 ESP12-E module

THis is my json code:
#define CONFIG_FILE  "/config.json"`
char * ssid;
const char * FILE_READ = "r";
void readcredentials() {
  if (!SPIFFS.begin()) {
    Serial.println("Failed to mount SPIFFS");
  } else {
    File file = SPIFFS.open(CONFIG_FILE, FILE_READ);

    if (!file) {
      Serial.println("There was an error opening the file");
      return;
    } else {
      Serial.println("File opened!");

      const size_t capacity = JSON_OBJECT_SIZE(8) + 150;
      DynamicJsonDocument doc(capacity);
      DeserializationError error = deserializeJson(doc, file);
      if (error) {
        Serial.println("error...");
        Serial.println(error.c_str());
      } else {
        strcpy(ssid, doc["mqtt_host"]);
      }
      Serial.println("");
    }
    file.close();
  }
}`

When i run this program i get IncompleteInput in the serial
I searched the web like hell buth dont find a solution, can someone please help me?

unbreakmat, 2020-09-14:
sorry for the messy code i tried it with the quotes buth something goes wrong :/

unbreakmat, 2020-09-14:
[image]

unbreakmat, 2020-09-14:
did some digging with this code:
`#define CONFIG_FILE "/config.json"
char * host;

void readcredentials() {
  if (!SPIFFS.begin()) {
    Serial.println("Failed to mount SPIFFS");
  } else {
    File file = SPIFFS.open(CONFIG_FILE, "r");

    if (!file) {
      Serial.println("There was an error opening the file");
      return;
    } else {
      Serial.println("File opened!");
      const size_t capacity = JSON_OBJECT_SIZE(1) + 20;
      DynamicJsonDocument doc(2048);
      deserializeJson(doc, file);
      serializeJson(doc, Serial);
      file.close();
    }
  }
}`

i actually see this in the serial: 00:16:55.551 -> {"mqtthost":"testhost"}
buth when i add this it crashes: 
`strcpy(host, doc["mqtthost"]);`

bblanchon, 2020-09-15:
Hi @unbreakmat,

Please check [the error code returned by deserializeJson()](https://arduinojson.org/v6/api/misc/deserializationerror/).

Best regards,
Benoit

unbreakmat, 2020-09-15:
dear blaanchon i already checked that link thats where i found the error.c_str() but i'm stuck now, i did some new research and found out this:
- i deserealise and it works with doc["somevalue"] == "somevalue" --> so i get values, the spifs works
- serealise (so writing new value) and the error says this error
- appending the doc and sending it back to the files with   for example object["objec"].set(_mqttServer); only saves this one, al the other object are "deleted" in my doc? i dont want this i only want to append the one i given, so no i just added them all but i dont like this way 
- i deserealise and it says again this error

my only conclusion would be that somehow in the deserealisation proces something went wrong and that the file becomes corrupted or by writing to it that the structure goes crippled of the file

any advice pleasy ? i attach in pastebin my currently code 
 https://pastebin.pl/view/ee18215c 

the config.json file is this now: 
`{"mqttserver":"some","mqttport":"1883","mqttuser":"noder","mqttpsk":"node","mqtttopic":"topic","host":"myesp","adminuser":"admin","adminpsk":"123"}`

best practise remove my objects in and leave 1, in the console you wil see in the last readspiffs that only 1 object was found, al the rest are deleted but are stil in the global vars i declared in the program on top
but i dont like that, it would be really great to append one in write that keeps the others

i already searched the arduinojsoon website with no luck :/

unbreakmat, 2020-09-15:
so my question is making the writesection append the file and let it work would be great
also the deserialsationerror on read and write working :)
if you have any advice in other could please let me know :)

bblanchon, 2020-09-16:
I'm sorry, but what is "writesection"?

If you want to append a JSON document to a file, just call `serializeJson(doc, file)` without clearing the file first.
In that case, you'll have to call `deserializeJson(doc, file)` repeatedly to read each JSON document in the file (*à la* [JSON Streaming](https://en.wikipedia.org/wiki/JSON_streaming))

unbreakmat, 2020-09-16:
with write section i meaned writing to the file, so do some changes, my bad of bad explenation ,sorry

about your appending solution:  just call serializeJson(doc, file) without clearing the file first. ==> am i clearing it in my section? (im fairly new to arduinojson sorry), can you point that code?

can you give me an example of the cod eyou mean because i dont get it :/

bblanchon, 2020-09-16:
`SPIFFS.open(path, "w")` truncates the file; you should use `"a"`  instead ([see doc](https://arduino-esp8266.readthedocs.io/en/latest/filesystem.html#open)).

If you're look for documentation, please check-out the "JSON streaming" section in the chapter "Advanced Techniques" of [Mastering ArduinoJson](https://arduinojson.org/book/).

unbreakmat, 2020-09-16:
thank you very much bblanchon , thats wat i needed i will figure it out and close this if it works :)

closed 2020-09-19 by unbreakmat