bblanchon/ArduinoJson #1380: To parse API (large Data)

ram-prasanth-9394, 2020-09-15:
Hi @bblanchon 

Can you help me to figure out this ?
Here i need to parse selected objects 
The Code is working fine but it returns Zero. 

The Sample output mentioned below,

**Code**

#define payload="https://disease.sh/v3/covid-19/countries"

JsonArray& root = jsonBuffer.parseArray(payload);
int sizeofArray =sizeof(root)/sizeof(int);
Serial.println("-----------------------------------");
Serial.println(sizeofArray);
for(int i=0;i<sizeofArray;i++){
JsonObject& root_0 = root[i];
long root_0_cases = root_0["cases"];
int root_0_deaths = root_0["deaths"];
int root_0_active = root_0["active"];
long root_0_recovered = root_0["recovered"];

Serial.print("Cases : ");
Serial.println(root_0_cases );
Serial.print("Deaths : ");
Serial.println(root_0_deaths );
Serial.print("Active : ");
Serial.println(root_0_active );
Serial.print("Recovered : ");
Serial.println(root_0_recovered );
}

**OUTPUT**

Cases : 0
Deaths : 0
Active : 0
Recovered : 0

bblanchon, 2020-09-16:
Hi @ram-prasanth-9394,

Please read:

* [Can I parse a JSON input that is too big to fit in memory?](https://arduinojson.org/v5/faq/can-i-parse-a-json-input-that-is-too-big-to-fit-in-memory/) (for ArduinoJson v5)
* [How to deserialize a very large document?](https://arduinojson.org/v6/how-to/deserialize-a-very-large-document/) (for ArduinoJson v6)

I strongly recommend upgrading to v6 because it uses less memory.

Best regards,
Benoit

ram-prasanth-9394, 2020-09-16:
@bblanchon 
can you plz give me an example im still getting Zero after changing Arduino to v6

bblanchon, 2020-09-19:
Hi @ram-prasanth-9394,

What error code is returned by [deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/)?

Best regards,
Benoit

ram-prasanth-9394, 2020-09-20:
#include <WiFi.h>
#include <HTTPClient.h> 
#include <ArduinoJson.h>
const char* ssid = "RideTARC";
const char* password = "987654321"; 
void setup () { 
Serial.begin(115200);
WiFi.begin(ssid, password); 
while (WiFi.status() != WL_CONNECTED) { 
delay(1000);
Serial.println("Connecting.."); 
} 
} 
void loop() {
if (WiFi.status() == WL_CONNECTED) {  
HTTPClient http; 
http.begin("https://disease.sh/v3/covid-19/countries");  
int httpCode = http.GET(); 
String payload = http.getString();
Serial.println(httpCode);
Serial.println(payload); 
const size_t capacity = JSON_ARRAY_SIZE(83) + 83*JSON_OBJECT_SIZE(8) + 7330;
DynamicJsonDocument doc(capacity);

deserializeJson(doc, payload);
int sizeofArray =sizeof(payload)/sizeof(int);
Serial.println("-----------------------------------");
Serial.println(sizeofArray);

    for(int i=0;i<sizeofArray;i++){
    JsonObject root_0 = doc[i];
    long root_0_cases = root_0["cases"];
    int root_0_deaths = root_0["deaths"];
    int root_0_active = root_0["active"];
    long root_0_recovered = root_0["recovered"];
    
    Serial.print("Cases : ");
    Serial.println(root_0_cases );
    Serial.print("Deaths : ");
    Serial.println(root_0_deaths );
    Serial.print("Active : ");
    Serial.println(root_0_active );
    Serial.print("Recovered : ");
    Serial.println(root_0_recovered );
}
 http.end();
 } 
 delay(3000);
}
 

**OUTPUT (SERIAL MONITOR)**

200  

-----------------------------------
3
Cases : 0
Deaths : 0
Active : 0
Recovered : 0
Cases : 0
Deaths : 0
Active : 0
Recovered : 0
Cases : 0
Deaths : 0
Active : 0
Recovered : 0

bblanchon, 2020-09-21:
Hi @ram-prasanth-9394,

You didn't check the error returned by [`deserializeJson()`](https://arduinojson.org/v6/api/json/deserializejson/), but I bet it's [`NoMemory`](https://arduinojson.org/v6/api/misc/deserializationerror/#nomemory) because the requested capacity is huge.

Your only solution is to implement the "deserialization in chunks" technique presented in [How to deserialize a very large document?](https://arduinojson.org/v6/how-to/deserialize-a-very-large-document/). You can find a complete example using this technique in the Reddit case study in the last chapter of [Mastering ArduinoJson](https://arduinojson.org/book/).

BTW, `sizeof(payload)/sizeof(int)` is not the right way to check the length of a `String`. Use `payload.size()` instead.

Best regards,
Benoit

closed 2020-09-30 by ram-prasanth-9394