bblanchon/ArduinoJson #1381: ArduinoStringAdapter.hpp:7:10: fatal error: WString.h: No such file or directory

youcangetme, 2020-09-16:
I am compiling for the new Teknic ClearCore using the Arduino wrappers and the Arduino IDE. It is pretty spot of Arduino compatible with a few exceptions. The architecture is described as "32-bit ARM Cortex M4F processor operating at 120MHz with 512kB FLASH and 192kB RAM. DSP and floating point instruction sets."

Here is the error I get

`
In file included from C:\Users\youca\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Strings/StringAdapters.hpp:16:0,
                 from C:\Users\youca\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Misc/SerializedValue.hpp:7,
                 from C:\Users\youca\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantData.hpp:8,
                 from C:\Users\youca\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/SlotFunctions.hpp:8,
                 from C:\Users\youca\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Array/ArrayIterator.hpp:7,
                 from C:\Users\youca\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Array/ArrayRef.hpp:8,
                 from C:\Users\youca\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:17,
                 from C:\Users\youca\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,
                 from C:\Users\youca\AppData\Local\Temp\arduino_modified_sketch_599557\JsonConfigFile.ino:22:
C:\Users\youca\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Strings/ArduinoStringAdapter.hpp:7:10: fatal error: WString.h: No such file or directory
 #include <WString.h>
          ^~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board Teknic ClearCore.`

I tried the usual workaround and included said libs directly in the same folder with no luck. This is just including the ArduinoJson.h so if I take it out all of these errors disappear.

I am sure this is pilot error and if somebody could give me direction as to what I am doing wrong it would be great. Thanks!

bblanchon, 2020-09-16:
Hi @youcangetme,

Thank you very much for reporting this issue.

Every Arduino-compatible core has a `WString.h` that contains the definition of the `String` class.
Unfortunately, Teknic ClearCore renamed this file to `ArduinoString.h`.

To me, this is a mistake from Teknic and they must align their file names to the official ones; otherwise, they cannot claim to be Arduino-compatible.
@youcangetme, can you contact Teknic support to report this issue?

Best regards,
Benoit

bblanchon, 2020-09-16:
I spoke too quickly.

I just discovered that Arduino team has [deprecated the `WString.h` header](https://github.com/arduino/ArduinoCore-API/blob/45e4e5aba6ad1a5b67026b686099cc0fce437cdc/api/deprecated/WString.h#L19):

> // including WString.h is deprecated, for all future projects use Arduino.h instead

I'll modify ArduinoJson to include `Arduino.h` instead.

closed 2020-09-16 by bblanchon

bblanchon, 2020-09-16:
The fix is available in the `6.x` branch and will be available in the next release.

However, there is still an error when compiling for Teknic ClearCore:

```
arduino/api/deprecated-avr-comp/avr/pgmspace.h:107:49: error: 'const void*' is not a pointer-to-object type
 #define pgm_read_ptr(addr) (*(const void *)(addr))
                                                 ^
```

This is due to [a bug in the hardware-independent core](https://github.com/arduino/ArduinoCore-API/pull/118).

Until this bug is fixed, you can either:

* [apply the change manually in `pgmspace.h`](https://github.com/arduino/ArduinoCore-API/pull/118/files#diff-7cf2fd59cb024f927f2c406eea1d7cc7), or
* [disable the support for PROGMEM in ArduinoJson](https://arduinojson.org/v6/api/config/enable_progmem/).

Thanks again for reporting this issue 👍

youcangetme, 2020-09-16:
Thanks @bblanchon ! This is progress so I am happy. I tried the other JSON libs and your's is the best license and the best code wise. Thank you for your time and help!

youcangetme, 2020-09-16:
@bblanchon I used the follow and got it to compile. Thank you for helping!

`#define ARDUINOJSON_ENABLE_PROGMEM 0
#include <ArduinoJson.h>`

commit 6a878ee4 (2020-09-16): Fixed error "No such file or directory WString.h" (fixes #1381)