bblanchon/ArduinoJson #1382: Give custom value if key not in the document?

usane1, 2020-09-16:
Hi!

I'm working on an ESP32 controlled neopixel LED project. I send JSON data to the ESP32 from an android app. I want to make multiple functions sended by separate JSON documents. I'm sendig tons of variables (RGBW color codes, LED by LED, and others).
So i have multiple JSON documents, but i dont want multiple procedures to deserialization so i decided to make a deserialization or all variable i can receive and check if the key is in the actual document or not. However if the key is not there i get zero for the actual variable BUT zero is a possible true value for lot of variables. For example
< my_var = doc["Temp"];> If Temp is not in the doc the my_var became 0, but 0 is acceptable for my_var, cannot use to indicate false input.
So i wonder if there is a way to make a deserialization parameter "if key is not there" or something that give  a customable value. For example "X" or false or anything i want instead of zero.

An other question: Can i acces  to the keys and values of the doc trough some arraylike way to be able to put them in a cycle?

bblanchon, 2020-09-17:
Hi @usane1,

You can find the answer to these questions in [the documentation](https://arduinojson.org/v6/doc/):

0. [JsonVariant::operator|](https://arduinojson.org/v6/api/jsonvariant/or/)
1. [JsonObject::begin() / end()](https://arduinojson.org/v6/api/jsonobject/begin_end/)
2. [copyArray()](https://arduinojson.org/v6/api/misc/copyarray/)

Please also have a look at the [Deserialization Tutorial](https://arduinojson.org/v6/doc/deserialization/).

Best regards,
Benoit

usane1, 2020-09-17:
Wow. Thanks you very much.
I looked the doc and the tutorial, but I missed out these somehow. I am bad at searching and finding.
Though i used this lib before, but just for regular JSON object.

Very amazing work, by the way. Thank you.

Regards,
Usane

bblanchon, 2020-09-17:
You're welcome, Usane.
Thank you for using ArduinoJson.
Don't hesitate to [star the library](https://github.com/bblanchon/ArduinoJson/stargazers) to show your support ;-)

usane1, 2020-09-17:
Star done.
However i still have problem with the copyArray. Mybe i want to use the wrong way, i don't know.
I need the JsonArray, or doc -> T[][] version.
Here is what i try:
The payload comes from websocket data.
<case WStype_TEXT:
{
  String message = String((char *)(payload));
  DynamicJsonDocument jdoc(1024);   
  DeserializationError err = deserializeJson(jdoc, message);
  Serial.println(err.c_str());
  JsonArray Temp = jdoc.to<JsonArray>();          
  copyArray(Temp, TestRGBW);              
}>
TestRGBW[2][4]   is declared on the beginning.
I got no error but after copy if i try to print the C array members to serial.
<Serial.println(Temp);
  for(i=0;i<2;i++)
  {
    for(uint8_t j=0;j<4;j++)
    {
      Serial.println(TestM[i][j]);
    }
  }>
And send for example {"RGBW": [[10, ,10, 10, 0], [20, 20, 20, 0]]}
I got all zero on the serial print.
What am i missing?

usane1, 2020-09-17:
Ok i partly solved it by the help of the assistant.
I used the doc.to<JsonArray>() to create the array, according to the example.
But the assistant dont put it to the doc. It works.
Question though, i have to know the array "key" name. Can i do it somehow to work with any key that arrives, or i have to check it myself?

bblanchon, 2020-09-18:
You can [enumerate key-value pairs](https://arduinojson.org/v6/api/jsonobject/begin_end/).

usane1, 2020-09-19:
Ok. Thank you very much.

closed 2020-09-19 by usane1