bblanchon/ArduinoJson #1383: Dynamic measuring whether JSON response contains an object / Number of elements of array

HamzaHajeir, 2020-09-16:
Hi

It's not an issue, It's just request for guidance, or could be a feature request if you decide it'll be needed.

I'm designing / implementing JSON request/response from and to the server.

consider this 'response' :

```json
{
  "cmd": {
    "system-activation": true, //Optional
    "pressure-mode": false,  //Optional
    "change-name": "Device0004", //Optional
    "change_timer": ["T01FF","T02FF"] //Optional
  }
}
```

The server 'could' issue commands, and could not be issuing 
and it 'could' send 1 element in "change_timer" array, or more than one. 

So, the API I found for command checking is : `doc.containsKey("cmd")` , Is it Ok for JSON Objects ?

About "change_timer" Array, Is there a method to calculate/measure how many elements inside it ? So I can dynamically process them one by one. (I'm not a fan of "check blindly for 5 elements ... "

Thanks

bblanchon, 2020-09-17:
Hi @HamzaHajeir,

Yes, there is a [JsonObject::containsKey()](https://arduinojson.org/v6/api/jsonobject/containskey/).

To get the number of elements in an array, use [JsonArray::size()](https://arduinojson.org/v6/api/jsonarray/size/).

Best regards,
Benoit

HamzaHajeir, 2020-09-20:
Great, Thanks @bblanchon .

closed 2020-09-20 by HamzaHajeir

bblanchon, 2020-09-21:
You're welcome, @HamzaHajeir.
Thank you for using ArduinoJson.