bblanchon/ArduinoJson #1384: Teknic ClearCore: deserializeJson() failed: NoMemory even when there appears to be enough RAM

youcangetme, 2020-09-17:
The Teknic Clear Core claims to have 192kB of RAM but when I run the following code I get out "NoMemory" error;

#define ARDUINOJSON_ENABLE_PROGMEM 0
#include <ArduinoJson.h>

  StaticJsonDocument<20> doc;
  DeserializationError error = deserializeJson(doc, file);
  if (error) {
    Serial.print("deserializeJson() failed: ");
    Serial.println(error.c_str());
    return;
  }

It doesn't matter what value I give it or how big or small the file is, the NoMemory error is always returned. So I ran the example code and no other libs loaded and I still get NoMemory. 

I think there is something besides "NoMemory" going on here but I could be wrong. One interesting thing to note is the output of the example JSON object to a file had no issues. It has no problem even with large objects created in memory and then serialized out to SD card.

If the stats are correct I am only using about 1/3rd the available RAM. So take that with a grain of salt.

"Sketch uses 211392 bytes (41%) of program storage space. Maximum is 507904 bytes.
Global variables use 43068 bytes of dynamic memory."

Thanks again for your help.

bblanchon, 2020-09-17:
Hi @youcangetme,

Please read:

* [Deserialization Tutorial](https://arduinojson.org/v6/doc/deserialization/)
* [How to determine the capacity of the JsonDocument?](https://arduinojson.org/v6/how-to/determine-the-capacity-of-the-jsondocument/)

Let me know if you have more questions.

Best regards,
Benoit

youcangetme, 2020-09-18:
It's a 25 char file. There is 150kbytes if RAM available using freemen lib for Arduino.

Very odd...

bblanchon, 2020-09-18:
In the snippet above, you chose a capacity of 20 bytes, which is probably insufficient.

youcangetme, 2020-09-18:
It doesn't even work with a file length of 1 char, for get 20 or 25! No combination of allocated memory or file size results in anything different. JSON support is gone. Configs are now a struct, zero memory issues with a struct of 200bytes. Maybe when I will revisit this later I will have time to go through the ArduiniJson code and find the issue but not right now. Thank you for your time.

closed 2020-09-18 by youcangetme