bblanchon/ArduinoJson #1385: serializeJSON sometimes returns garbage

davidkjackson54, 2020-09-18:
Using ArduinoJSON in PlatformIO on ESP32
I define 
std::string deviceAddress = foundDevices.getDevice(i).getAddress().toString();  
and occasionally - maybe 1 time in 10 I get one field that hols hte above value shown as garbage.

If I change the above statement to 
String deviceAddress = foundDevices.getDevice(i).getAddress().toString().c_str();
When I make this change I (so far) have been unable to recreate the field displayed as garbage.

Any clues as to why it occurs when I define device Address as std:string? 

The output display looks like this when I run it in a loop: notice the garbage in the mac address field on the second occurrence. 
Scanning...
d2:fe:0e:a1:90:45
Sending..
{"tilts":[{"colour":7,"mac":"**bd21a77601762402596090**","uuid":"a495bb70-c5b1-4b44-b512-1370f02d74de","temp":24.44444,"gravity":1.005,"rssi":-60}]}
Message arrived on topic: scan/tilts Message: {}
Scanning...
d2:fe:0e:a1:90:45
Sending..
{"tilts":[{"colour":7,"mac":"d2:fe:0e:a1:90:45","uuid":"a495bb70-c5b1-4b44-b512-1370f02d74de","temp":24.44444,"gravity":1.005,"rssi":-66}]}
Message arrived on topic: scan/tilts Message: {}

The code:
`const size_t capacity = JSON_ARRAY_SIZE(8) + JSON_OBJECT_SIZE(1) + 8*JSON_OBJECT_SIZE(6) +430;
    DynamicJsonDocument outdoc(capacity); 
    JsonArray devices = outdoc.createNestedArray("tilts");

    for (uint32_t i = 0; i < m_deviceCount; i++)  {
      BLEAdvertisedDevice Device = foundDevices.getDevice(i);
                  
      String devString = Device.toString().c_str();
      String devData = devString.substring(63);                              // a495bb70c5b14b44b5121370f02d74de004c0424c5 
      if (devData.substring(7, 32) == "0c5b14b44b5121370f02d74de") {        // Tilt found

        //String deviceAddress = foundDevices.getDevice(i).getAddress().toString().c_str();
        std::string deviceAddress = foundDevices.getDevice(i).getAddress().toString();  // get MAC address
        int deviceRSSI = foundDevices.getDevice(i).getRSSI();                           // get RSSI         
        String x = devices[m_tiltCount]["mac"];                           // check if a field is already filled
        if (x == "null") {
          devices[m_tiltCount]["colour"] = parseColour(devData);          // extract Tilt Colour offset e.g. Yellow = 7
          //devices[m_tiltCount]["mac"] = deviceAddress;  
          devices[m_tiltCount]["mac"] = deviceAddress.c_str();            // MAC address
          **Serial.println(deviceAddress.c_str());**
          devices[m_tiltCount]["uuid"] = parseUUID(devData);              // extract UUID
          devices[m_tiltCount]["temp"] = parseTemp(devData);              // extract temperature in C
          devices[m_tiltCount]["gravity"] = parseGravity(devData);        // extract gravity
          devices[m_tiltCount]["rssi"] = deviceRSSI;                      // RSSI
           m_tiltCount++;
        }
     }
    } 
    char buffer[1200];
    if (m_tiltCount) {
    //serializeJson(outdoc, Serial);                  // just tempo for testing
    
      serializeJson(outdoc, buffer);
      Serial.println("Sending..");
      Serial.println(buffer);  
      client.setBufferSize(1200);
      client.publish("scan/tilts/done", buffer); 
      client.setBufferSize(256);                    // reset to default
`

Please provide all the relevant information:
* good title
* short description of the problem
* target platform
* compiler model and version
* MVCE (https://stackoverflow.com/help/mcve)
* compiler output

Good questions get fast answers!
-->

bblanchon, 2020-09-19:
Hi @davidkjackson54,

Please read:

* [Why does the output contain garbage?](https://arduinojson.org/v6/faq/why-does-the-output-contain-garbage/)

Best regards,
Benoit

davidkjackson54, 2020-09-19:
Thanks. I had resolved the issue myself but having an explanation is a great help. I hadn't found that link till you pointed it out.

closed 2020-09-19 by davidkjackson54