bblanchon/ArduinoJson #1386: Memory Allocation Fails on ATMEL ATSAMD51N19A chips

youcangetme, 2020-09-19:
I can pretty much assure you memory allocation in ArduinoJson is failing on Atmel (Microchip) ATSAMD51N19A chips.

I am not sure what the problem is but it "looks like" the memory allocation always returns 0 bytes. I tried to debug it myself but I am truly late on my delivery. The binary config was a bridge too far for my client which is why I revisited it.

The Arduino.cc "beta" Arduino_Json library has no such issue on these chips with the exact same file, no changes. The biggest different is you pass Arduino_Json a char array to parse, it doesn't allocate the memory itself.

Best of luck!

https://www.microchip.com/wwwproducts/en/ATSAMD51N19A

bblanchon, 2020-09-20:
Hi @youcangetme,

That's really surprising.
Are the examples working?

Best regards,
Benoit

bblanchon, 2020-09-21:
@youcangetme, I'm sorry to stress you, but can you confirm that the examples don't work?

I really doubt this is possible, but if it's true, I need to get my hands on one of these devices immediately.
I regularly test on SAMD21, but I don't own a SAMD51.
Can you recommend an inexpensive board that embeds this microcontroller?

youcangetme, 2020-09-22:
I did test the examples and the ones I tried failed. Adafruit sells one (Grand Central Express) but It might be a different chip within the D51s. In fact I have one laying around here somewhere. I will pull it out and see if I can get it to replicate the issue and get back to you. They run about $40USD whereas the ClearCore are hardened industrial 24 volt I/O Arduinos and run $150.

bblanchon, 2020-09-22:
I went for an [Adafruit ItsyBitsy M4 Express](https://amzn.to/2ZZkMFU); it was the cheapest I could find.

bblanchon, 2020-09-23:
Hi @youcangetme,

I received the [Adafruit ItsyBitsy M4 Express](https://amzn.to/2ZZkMFU).
I can confirm that the following examples are working as expected:

* [JsonParserExample.ino](https://arduinojson.org/v6/example/parser/)
* [JsonFilterExample.ino](https://arduinojson.org/v6/example/filter)
* [JsonGeneratorExample.ino](https://arduinojson.org/v6/example/generator/)

It's a SAMD51, like the ClearCore, but I'm not using the same Arduino core.
I tested with the [Adafruit SAMD Arduino Core](https://github.com/adafruit/ArduinoCore-samd), whereas you're using the [ClearCore Arduino Wrapper](https://github.com/Teknic-Inc/ClearCore-Arduino-wrapper).
This means the problem could be related to the core.

Can you double-check the example, and see if you can find out more about the issue?

BTW, it's the first time someone tells me that Arduino_JSON works better than ArduinoJson; usually, it's the other way around.
After re-reading your message I realized that ArduinoJson is not shown on arduino.cc.
I don't know how this is possible, since other JSON libraries and my other libraries are showing.
I suspect that they banned ArduinoJson to promote their own library, so I tried to [contact them on twitter](https://twitter.com/BenoitBlanchon/status/1308850178577895430).

Best regards,
Benoit

bblanchon, 2020-10-16:
Hi @youcangetme,

Did you get the opportunity to check the examples again?

Best regards,
Benoit

youcangetme, 2020-10-16:
I would love to help you. I am behind schedule getting this stupid project shrink wrapped and out the door though. Deaths by a thousand cuts? When I have this beast at v1 I will circle back around. I think ArduinoJson has better type conversion when assigning values. The other package certainly needs more "hints" to figure what you are asking if it. That I will grant you.

closed 2020-11-01 by bblanchon