bblanchon/ArduinoJson #1387: delete and shift jsonArray

m2sarmadi, 2020-09-21:
hello everyone
I have an Json array similar to the following :
`[{"RoomId":0,"RoomType":"LivingRoom"},{"RoomId":1,"RoomType":"BedRoom"},{"RoomId":2,"RoomType":"Kitchen"}]`

I want to delete one particular object and shift the rest of array so that the "RoomId" are in order.
for example : if i delete object with "RoomId=1" , So the array becomes like this :
`[{"RoomId":0,"RoomType":"LivingRoom"},{"RoomId":1,"RoomType":"Kitchen"}]`

I can delete an object of array using this code :
```
for(Count=0; Count < SizeOfArray ; Count++) {
      if (array[Count]["RoomId"] == 1) {
        array.remove(1);
  }
```
But I can't shift the rest of the array.

bblanchon, 2020-09-21:
Hi @m2sarmadi,

You cannot reorder the elements of a [JsonArray](https://arduinojson.org/v6/api/jsonarray/); you need to recreate a new array.

However, removing an element matching criteria (and preserving the order) can be easily implemented with iterators:

```c++
for (JsonArray::iterator it=arr.begin(); it!=arr.end(); ++it) {
  if ((*it)["RoomId"] == 1)
      arr.remove(it);
}
```

See online demo: https://wandbox.org/permlink/7hwvQROY4UGtQrIm

Best regards,
Benoit

ewaldc, 2020-09-21:
Assuming _n_ is the element to delete, how about:
```
    bool renumber = false;
    int count = -1;
    for (JsonArray::iterator it = arr.begin(); it != arr.end(); ++it) {
      if ((*it)["RoomId"] == n) {
        arr.remove(it);
        renumber = true;
      } else count++;
      if (renumber) (*it)["RoomId"] = count;
    }
```

closed 2020-10-03 by m2sarmadi