bblanchon/ArduinoJson #1388: Large API Call Not Parsing

OlivierLdeB, 2020-09-23:
Hi,

I am using the ESP8266 NodeMCU v3.0 with ArduinoJson version 6.16.1, and I'm having issues with the new improved filtering. I'm working with the [NHL's boxscore API](https://statsapi.web.nhl.com/api/v1/game/2017020653/boxscore). Since the API response is quite big, I've decided to use a filter document to keep only the relevant data. 

Here's an example program of what I've tried. I've had success with this method in the past with smaller APIs.

`#include <ArduinoJson.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266WiFi.h>

//wifi credentials
#define ssid "xxxxx"
#define pswd "xxxxxxxxxxxxx"

void setup() {
  Serial.begin(115200);
  Serial.println();

  WiFi.begin(ssid, pswd);

  Serial.print("Connecting");
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
  Serial.println();

  Serial.print("Connected, IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  HTTPClient http;
  http.begin("https://statsapi.web.nhl.com/api/v1/game/2017020653/boxscore", "a7996d27d8c805fd2d6e97587347b204c5bee21e");
  int httpCode = http.GET();
  Serial.println(httpCode);

  DynamicJsonDocument filter(800);
  filter["teams"]["away"]["team"]["name"] = true;
  filter["teams"]["away"]["team"]["id"] = true;
  filter["teams"]["away"]["teamStats"]["teamSkaterStats"]["goals"] = true;
  filter["teams"]["away"]["teamStats"]["teamSkaterStats"]["shots"] = true;

  filter["teams"]["home"]["team"]["name"] = true;
  filter["teams"]["home"]["team"]["id"] = true;
  filter["teams"]["home"]["teamStats"]["teamSkaterStats"]["goals"] = true;
  filter["teams"]["home"]["teamStats"]["teamSkaterStats"]["shots"] = true;

  DynamicJsonDocument doc(2000);
  deserializeJson(doc, http.getStream(), DeserializationOption::Filter(filter));

  Serial.println(doc["teams"]["away"]["team"]["name"].as<char*>());
  Serial.println(doc["teams"]["away"]["team"]["id"].as<int>());
  Serial.println(doc["teams"]["away"]["teamStats"]["teamSkaterStats"]["goals"].as<int>());
  Serial.println(doc["teams"]["away"]["teamStats"]["teamSkaterStats"]["shots"].as<int>());

  Serial.println(doc["teams"]["home"]["team"]["name"].as<char*>());
  Serial.println(doc["teams"]["home"]["team"]["id"].as<int>());
  Serial.println(doc["teams"]["home"]["teamStats"]["teamSkaterStats"]["goals"].as<int>());
  Serial.println(doc["teams"]["home"]["teamStats"]["teamSkaterStats"]["shots"].as<int>());

  http.end();
  delay(60000);
}`

The output I get leads me to believe the API call wasn't parsed. In the past this has been because the response was too large. It was my understanding that filtering was supposed to fix that issue. 

`Connecting.....
Connected, IP address: 192.168.1.219
200

0
0
0

0
0
0
null`

I hope I've properly laid out my problem, and thanks in advance for your time and help!

Olivier Larose-de Billy

OlivierLdeB, 2020-09-23:
Sorry, I thought the 'insert code' option would format my code, but attached is 
my code and a screenshot of the serial response.

[filterBoxscore.docx](https://github.com/bblanchon/ArduinoJson/files/5265115/filterBoxscore.docx)

[image]

bblanchon, 2020-09-23:
Hi Olivier,

Please check the error code returned by [deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/); it will tell you why the results are null.
BTW, I noticed that you used `getStream()`, which is good, but you forgot to switch to HTTP/1.0 to disable [Chunk Transfer Encoding](https://en.wikipedia.org/wiki/Chunked_transfer_encoding).
Also, 2000 bytes may not be enough, did you check with the [ArduinoJson Assistant](https://arduinojson.org/v6/assistant/).

See also:

* [How to use ArduinoJson with HTTPClient?](https://arduinojson.org/v6/how-to/use-arduinojson-with-httpclient/)
* [How to deserialize a very large document?](https://arduinojson.org/v6/how-to/deserialize-a-very-large-document/)

Best regards,
Benoit

OlivierLdeB, 2020-09-23:
The call for HTTP version 1.0 was what was missing. I'll also keep deserializeJson() in mind. 
Thanks a ton for your fast and clear response. I love your library!

Olivier Larose-de Billy

closed 2020-09-23 by OlivierLdeB

bblanchon, 2020-09-24:
You're welcome, Olivier.
Thank you for using ArduinoJson.