bblanchon/ArduinoJson #1389: Parsing the key seems to fail

Gilbert54, 2020-09-25:
I'm using ESP32 with Arduino1.8.13 and ArduinoJSON 6.16.1.

For testing purposes I create JSON string "{\"CMD\":\"PIN\",\"PIN\":\"123456\"}".
Then I check the command "CMD" like this:

  char command[5] = {jsonDoc["CMD"]};
  if(strcmp_P(command, PSTR("PIN")))
    verifyPin(jsonDoc["PIN"]);    
  else if(strcmp_P(command, PSTR("GET")))
    parseGetCommand(jsonDoc);    
  else if (strcmp_P(command, PSTR("SET")))
    parseSetCommand(jsonDoc);    
  else
    returnCmdError(501);

This works as expected. The comparison strcmp_P(command, PSTR("PIN")) results in "true" and the PIN value is verified.

Now I play the bad guy and modify my JSON string to "{\"ERR\":\"PIN\",\"PIN\":\"123456\"}".
I would expect that char command[5] = {jsonDoc["CMD"]} produces null or an error since jsonDoc["CMD"] doesn't exist,
but the program still gets the PIN. Am I missing something here?
If I do the same with element "PIN" (i.e. change it to "PIP", I get what I expected: jsonDoc["PIN"] = null.

[ArduinoCode.ino.txt](https://github.com/bblanchon/ArduinoJson/files/5282032/ArduinoCode.ino.txt)

bblanchon, 2020-09-26:
Hi @Gilbert54,

First, [ESP32 doesn't need PSTR()](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/pgmspace.h#L35), you can get rid of it.

Second, `strcmp()` returns 0 when the strings match, so you should write `if (!strcmp(command, "PIN"))`.

Third, when you write `char command[5] = {jsonDoc["CMD"]};`, the compiler sees `char command[5] = { (int)jsonDoc["CMD"] };`.
So, in practice, it does `char command[5] = {0};`.
Instead you should write `const char* command = jsonDoc["CMD"];`.

Forth, [JsonVariant](https://arduinojson.org/v6/api/jsonvariant/) has an `==` operator that supports string comparisons, you can write `if (jsonDoc["CMD"] == "PIN")`.

Finally, here is how you can rewrite this snippet.

```c++
JsonVariantConst cmd = jsonDoc["CMD"];
if (cmd == "PIN")
  verifyPin(jsonDoc["PIN"]);    
else if (cmd == "GET")
  parseGetCommand(jsonDoc);    
else if (cmd == "SET")
  parseSetCommand(jsonDoc);    
else
  returnCmdError(501);
```

@Gilbert54, please take some time to explore the [documentation on arduinojson.org](https://arduinojson.org/v6/doc/).
If you haven't done it already, please follow the [Deserialization Tutorial](https://arduinojson.org/v6/doc/deserialization/).
If you need a refresher on strings and stuff like that, please check out "The Missing C++ Course" in [Mastering ArduinoJson](https://arduinojson.org/book/).

Best regards,
Benoit

Gilbert54, 2020-09-26:
Benoît,

thanks for the quick and helpful response. Meanwhile I sorted out my string
issues. Coming from C# handling strings in Arduino C++ can be challenging.
The JsonVariant == is a  nice simplification.
I changed my code according to your recommendations. When I test this it
produces the expected result. However, when I replace the first Key "CMD"
by something else (e.g. "ERR") the code fails in a nasty way. It appears to
send "nothing found" to the Serial port, but my program can't catch this
error. Any suggestions?

PS: why is PSTR() not needed on ESP32? There are plenty of articles on the
web describing how to save RAM on ESP devices, all involving PSTR or
similar techniques.

************************************
Code:
...
 const char * jsonString = "{\"ERR\":\"PIN\",\"PIN\":\"123456\"}";

  JsonVariantConst command = jsonDoc["CMD"];
  //const char * command = jsonDoc["CMD"];
  //String command = jsonDoc["CMD"];

  Serial.print("COMMAND = ");
  Serial.println(command.as<char*>());            // see
https://arduinojson.org/v6/api/jsonarrayconst/begin_end/

  if(command == "PIN"){
    Serial.println("PIN command");
    verifyPin(jsonDoc["PIN"]);
  }

... etc...

***************************************
Serial output:

13:44:59.659 -> ets Jun  8 2016 00:22:57
13:44:59.659 ->
13:44:59.659 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
13:44:59.659 -> configsip: 0, SPIWP:0xee
13:44:59.659 ->
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
13:44:59.659 -> mode:DIO, clock div:1
13:44:59.659 -> load:0x3fff0018,len:4
13:44:59.692 -> load:0x3fff001c,len:1044
13:44:59.692 -> load:0x40078000,len:8896
13:44:59.692 -> load:0x40080400,len:5816
13:44:59.692 -> entry 0x400806ac
13:44:59.794 -> DeserializationError = Ok
13:44:59.794 -> COMMAND =
13:44:59.794 -> nothing found
13:44:59.794 -> he following tasks did not reset the watchdog in time:
13:44:59.794 ->

13:44:59.794 -> COMMAND =   => the "CMD" key doesn't exist
and jsonDoc["CMD"] produces 'nothing' (I suppose it really produces null?).
That's where my code appears to be losing control.

Kind regards,
Gilbert

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virusvrij.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Gilbert54, 2020-09-26:
I found a solution, maybe not the smartest, but it does the job for me.

  JsonObject root = jsonDoc.as<JsonObject>();
  if (!root.containsKey("CMD")){
    Serial.println("CMD missing!");
    returnCmdError(400);                        // 400 Bad Request
    return;

Before using a Key, I can check if it exists.A more robust and simple
solution could be that retrieving a value for a no-existing Key would
return null?

The reason why I'm so pedantic about the possibility that a Key may be
missing is that the ESP device operates in a noisy environment (on a
motorcycle, mounted near the ECU). Apart from programming errors on the
client side (Android App via Bluetooth) there is the possibility that
transmission could be compromised by noise signals produced by the
motorcycle's electronics.

Kind regards,
Gilbert

---------- Forwarded message ---------
Van: Gilbert Hersschens <ghersschens@gmail.com>
Date: za 26 sep. 2020 om 14:00
Subject: Re: [bblanchon/ArduinoJson] Parsing the key seems to fail (#1389)
To: bblanchon/ArduinoJson <
reply@reply.github.com>

Benoît,

thanks for the quick and helpful response. Meanwhile I sorted out my string
issues. Coming from C# handling strings in Arduino C++ can be challenging.
The JsonVariant == is a  nice simplification.
I changed my code according to your recommendations. When I test this it
produces the expected result. However, when I replace the first Key "CMD"
by something else (e.g. "ERR") the code fails in a nasty way. It appears to
send "nothing found" to the Serial port, but my program can't catch this
error. Any suggestions?

PS: why is PSTR() not needed on ESP32? There are plenty of articles on the
web describing how to save RAM on ESP devices, all involving PSTR or
similar techniques.

************************************
Code:
...
 const char * jsonString = "{\"ERR\":\"PIN\",\"PIN\":\"123456\"}";

  JsonVariantConst command = jsonDoc["CMD"];
  //const char * command = jsonDoc["CMD"];
  //String command = jsonDoc["CMD"];

  Serial.print("COMMAND = ");
  Serial.println(command.as<char*>());            // see
https://arduinojson.org/v6/api/jsonarrayconst/begin_end/

  if(command == "PIN"){
    Serial.println("PIN command");
    verifyPin(jsonDoc["PIN"]);
  }

... etc...

***************************************
Serial output:

13:44:59.659 -> ets Jun  8 2016 00:22:57
13:44:59.659 ->
13:44:59.659 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
13:44:59.659 -> configsip: 0, SPIWP:0xee
13:44:59.659 ->
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
13:44:59.659 -> mode:DIO, clock div:1
13:44:59.659 -> load:0x3fff0018,len:4
13:44:59.692 -> load:0x3fff001c,len:1044
13:44:59.692 -> load:0x40078000,len:8896
13:44:59.692 -> load:0x40080400,len:5816
13:44:59.692 -> entry 0x400806ac
13:44:59.794 -> DeserializationError = Ok
13:44:59.794 -> COMMAND =
13:44:59.794 -> nothing found
13:44:59.794 -> he following tasks did not reset the watchdog in time:
13:44:59.794 ->

13:44:59.794 -> COMMAND =   => the "CMD" key doesn't exist
and jsonDoc["CMD"] produces 'nothing' (I suppose it really produces null?).
That's where my code appears to be losing control.

Kind regards,
Gilbert

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virusvrij.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#m_8975364692149815933_DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

bblanchon, 2020-09-28:
Hi Gilbert,

> A more robust and simple solution could be that retrieving a value for a no-existing Key would
return null?

That's precisely what it does.
If not, please provide a reproduction code so that I can fix it immediately.

Best regards,
Benoit

Gilbert54, 2020-09-28:
Thanks for your help, Benoît. While I was staring at JsonDocument I didn't
see JsonVariant. Variant has all the bells and whistles I need to handle
the bad stuff.

Kind regards,
Gilbert

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virusvrij.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

closed 2020-09-28 by bblanchon