bblanchon/ArduinoJson #1390: Get path from jsonobject

Findus76, 2020-09-28:
Hello

Writes a function to check incoming json against own protocol.
The protocol must contain certain keys, eg "Info", "R / W", "Settings", etc.
For example, the json message may look like this.
`
{
	"Type": "SystemFunction",
	"Function list": {
		"Slot 1": {
			"Settings": {},
			"Info": {
				"Position": "Mobile",
				"Description": "1,5V max 4A, A, AA, AAA",
				"Battery Type": "",
				"Battery ID": 0,
				"State": "",
				"Time(s)": 0},
			"Voltage":{
				"Head Type": "Sensor",
				"Sub Type": "Voltage",
				"R/W": "R",
				"Unit": "V",
				"Value": 0},
			"Current":{
				"Head Type": "Sensor",
				"Sub Type": "Voltage",
				"R/W": "R",
				"Unit": "A",
				"Value": 0},
			"Total Charge":{
				"Head Type": "Sensor",
				"Sub Type": "Ampere hour",
				"R/W": "R",
				"Unit": "mAh",
				"Value": 0},
			"Total Discharge":{
				"Head Type": "Sensor",
				"Sub Type": "Ampere hour",
				"R/W": "R",
				"Unit": "mAh",
				"Value": 0}}}}
`

I would like to write a function that takes arguments jsonobject.
For example (above), reference to key "Voltage". If that function sees that key "R / W" is missing, I would like to print an error message "Key: "R / W" is missing under  \ Function list \ Slot 1 \ Voltage ".
I dont find any way to find the path from a jsonobject.

Brg

bblanchon, 2020-09-29:
Hi @Findus76,

Given a `JsonDocument` and a `JsonObject`, there is no way to get the "path" to the `JsonObject` in the `JsonDocument`.
In fact, there is no notion of "path" in ArduinoJson.
There is a feature request for a JSONPath resolution (#821), but it's not implemented yet, and it would only go forward.

To work around this limitation, you'll have to handle the path in your program: use a string to remember which path you followed to get to this `JsonObject`.

Best regards,
Benoit

Findus76, 2020-09-29:
Hi Benoit

Thanks.

I will handle it in my program.

Brg

closed 2020-09-29 by Findus76