bblanchon/ArduinoJson #1391: Case independent key selection

matswebjorn, 2020-09-29:
Is there some way to configure ArduinoJson so that it becomes case independent when extracting values. 
I.e assume a key "firstLady" that I'd like to extract using
`const char* name = doc["firstlady"];`

bblanchon, 2020-09-30:
Hi @matswebjorn,

ArduinoJson doesn't natively support case-insensitive lookup.

However, you could use an `std::string` with custom character traits.
See here for an example: https://stackoverflow.com/a/5319855/1164966

If `std::string` is not an option, then you can use [iterators](https://arduinojson.org/v6/api/jsonobject/begin_end/) to find the key.

Best regards,
Benoit

matswebjorn, 2020-10-01:
Hi Benoit,

Thanks, I’ll try your suggestions

Regards
Mats

closed 2020-10-16 by bblanchon