bblanchon/ArduinoJson #1392: Setting nested object fails for object set with returned value

torqu3e, 2020-09-30:
If a JsonDocument is set via a value returned from another function then setting nested objects does not work and fails silently. On the other hand if the same JsonDocument is set directly via a text string the nested objects get set correctly.

This is strange behaviour and I've not been able to find anything about based on my reading. Is it something I am doing wrong or something amiss in the library?

Target platform - ESP8266 (Lolin Wemos D1 mini) Arduinjson 6.16.1 on Arduino 1.8.13

Tests via HTTP server -

Test when Jsondocument is set directly using the string, nested object gets value set

```
$ curl http://192.168.1.121/test
{
  "msg": "directly set text",
  "state": {
    "a": "0",
    "b": "1"
  }
}
```

Test when Jsondocument is set from another function, nested object does not get set

```
$ curl http://192.168.1.121/test
{
  "msg": "text from function",
  "state": {}
}
```

Serial console output for both above tests together
```
HTTP server started on ajtest.local at ip 192.168.1.121
{
  "msg": "directly set text",
  "state": {
    "a": "0",
    "b": "1"
  }
}

HTTP server started on ajtest.local at ip 192.168.1.121
{
  "msg": "text from function",
  "state": {}
}
```

Sample code to reproduce problem, only change between the two tests is commenting one or the other `doc["msg"]` line, and re-uploading

```
#include <ArduinoJson.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>

#define HOSTNAME "ajtest"

WiFiManager wifiManager;
ESP8266WebServer server(80);

void setup() {

  Serial.begin(115200);

  wifiManager.autoConnect("test_AP");

  server.on("/test", HTTP_GET, test_func);
  server.begin();

  Serial.print("HTTP server started on ");
  Serial.print(HOSTNAME);
  Serial.print(".local");
  Serial.println(" at ip " + WiFi.localIP().toString());
}

void test_func() {
  String message = "";
  const size_t capacity = 2*JSON_OBJECT_SIZE(2);
  DynamicJsonDocument doc(capacity);

//  doc["msg"] = "directly set text";
  doc["msg"] = ret_func();

  JsonObject state = doc.createNestedObject("state");
  state["a"] = "0";
  state["b"] = "1";

  serializeJsonPretty(doc, message);
  Serial.println(message);
  server.send(200, "application/json", message);
}

String ret_func() {
    return ("text from function");
}

void loop() {
  server.handleClient();
}
```
Compiler output
```
Executable segment sizes:
IROM   : 279960          - code in flash         (default or ICACHE_FLASH_ATTR) 
IRAM   : 27212   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 
DATA   : 1328  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 1344  ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 29712 )         - zeroed variables      (global, static) in RAM/HEAP 
Sketch uses 309844 bytes (29%) of program storage space. Maximum is 1044464 bytes.
Global variables use 32384 bytes (39%) of dynamic memory, leaving 49536 bytes for local variables. Maximum is 81920 bytes.
esptool.py v2.8
Serial port /dev/cu.SLAB_USBtoUART
Connecting........_
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: ec:fa:bc:5e:f7:d6
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0240
Compressed 314000 bytes to 225732...
Wrote 314000 bytes (225732 compressed) at 0x00000000 in 5.4 seconds (effective 466.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

bblanchon, 2020-09-30:
Hi @torqu3e,

When you insert a `String` (as opposed to a `const char*`), you must increase the capacity of the `JsonDocument` because ArduinoJson makes a copy of the string.

It's one of the core principles of ArduinoJson; I don't know why you didn't read about it earlier, since it's described in:

* [Serialization Tutorial](https://arduinojson.org/v6/doc/serialization/), section 4.6: "Duplication of strings"
* [How to determine the capacity of the JsonDocument?](https://arduinojson.org/v6/how-to/determine-the-capacity-of-the-jsondocument/)
* [Why is the output incomplete?](https://arduinojson.org/v6/faq/why-is-the-output-incomplete/)
* [StringExample.ino](https://arduinojson.org/v6/example/string/)

There is probably a lot of room for improvement in the documentation.
Please let me know if you have any suggestions.
For example, where did you search the information, and why did you miss the existing one?

Best regards,
Benoit

torqu3e, 2020-10-01:
@bblanchon Thank you so much for the prompt response. I had a vague idea it was to do with `String` vs `const char *` but not the specifics. I skimmed through the Serialization tutorial and it was on my deep dive list though missing the FAQ is completely on me.

Guess why I got into this situation was using the assistant to get capacity. The Serialization example gives 
`const size_t capacity = 2*JSON_OBJECT_SIZE(2);`
vs Deserialization gives 
`const size_t capacity = 2*JSON_OBJECT_SIZE(2) + 40;`

and blindly copying that, expecting it to work. Just adding the capacity (+ 40) for the text string resolves the issue.

Really appreciate the help and your effort on writing a great library! 👍

bblanchon, 2020-10-01:
You're welcome, @torqu3e.
Thank you for using ArduinoJson!
Don't hesitate to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project 😉

closed 2020-10-01 by bblanchon