bblanchon/ArduinoJson #1393: Getting values from multiple objects!

masonhol, 2020-10-01:
Hi, I am trying to read in the values of the "sensor" from the Json text. I am able to do this for the first object in the code shown below. But how can I do this if there is more than one obect and sensor value? 

As shown in the code below i want to be able to read in the sensor value "gps1" AND "gps2"...

#include <ArduinoJson.h>

void setup() {
  // Initialize serial port
  Serial.begin(9600);
  while (!Serial) continue;

  DynamicJsonDocument doc(200);

  char json[] = "{\"sensor\":\"gps1\",\"time\":1351824120,\"data\":[48.756080,2.302038]},{\"sensor\":\"gps2\",\"time\":1351824120,\"data\":[48.756080,2.302038]}";
      Serial.println(json);

  // Deserialize the JSON document
  DeserializationError error = deserializeJson(doc, json);

  // Test if parsing succeeds.
  if (error) {
    Serial.print(F("deserializeJson() failed: "));
    Serial.println(error.c_str());
    return;
  }

    String sensor = doc["sensor"];
    // Print values.
    Serial.println(sensor);
  }

void loop() {
  
}

bblanchon, 2020-10-01:
Hi @masonhol,

If there are a few objects, you can put them in an array, like so:

```json
[{"sensor":"gps1","time":1351824120,"data":[48.756080,2.302038]},{"sensor":"gps2","time":1351824120,"data":[48.756080,2.302038]}]
```

Then, you can use [iterators](https://arduinojson.org/v6/api/jsonarray/begin_end/) to enumerate all values.
Please have a look at the [Deserialization Tutorial](https://arduinojson.org/v6/doc/deserialization/) for details.

However, if there are many objects, you should use a different technique called "JSON Streaming."
In this case,  the JSON document must come from a stream (i.e., not from a `char[]`), and you must call [`deserializeJson()`](https://arduinojson.org/v6/api/json/deserializejson/) repeatedly.
You can learn more in [Mastering ArduinoJson](https://arduinojson.org/book/), chapter "Advanced Techniques."

Best regards,
Benoit

masonhol, 2020-10-01:
> Hi @masonhol,
>
> If there are a few objects, you can put them in an array, like so:
>
> ```json [...]

Thanks for the reply.
The project involves sending a Json formatted string to an arduino via serial connection. This string will contain anywhere form 10 to 25 objects, each containing 6 elements which need to be extracted.

What do you suggest is the best method for this?

Many thanks.

bblanchon, 2020-10-02:
This sounds like something that can fit in RAM, so you can try the array technique.

See also: [How to do serial communication between two boards?](https://arduinojson.org/v6/how-to/do-serial-communication-between-two-boards/)

masonhol, 2020-10-12:
Hi, thanks for your help. I have another question though.

I am trying to use "Json streaming" to read the serialised string sent from
a nodemcu to an arduino uno through a serial connection.
The string i am sending looks like this, (but may contain many more
objects);

String input = "[{\"Name\":\"WW
6-6\",\"Freq\":\"0\",\"Day\":\"Monday\",\"Start_time\":\"18:00\",\"Date\":null},{\"Name\":\"Forbes\",\"Freq\":\"0\",\"Day\":\"All\",\"Start_time\":\"10:30\",\"Date\":null},{\"Name\":\"LCCC
Training\",\"Freq\":\"0\",\"Day\":\"Monday\",\"Start_time\":\"5:15\",\"Date\":null}]";

I serialized the string on the nodemcu and sent it over serial to the
Arduino.

This is my code for the Arduino:

#include <ArduinoJson.h>

void setup() {
  // Initialize "debug" serial port
  // The data rate must be much higher than the "link" serial port
  Serial.begin(115200);
  while (!Serial) continue;

}

void loop() {
  // Check if the other Arduino is transmitting

if (Serial.available())
  {
        // Jump inside the array
      Serial.find("[");
      do {
          // Deserialize the next post
           auto err = deserializeJson(doc, Serial);
           if (err) {
            Serial.print(F("deserializeJson() failed with code "));
            Serial.println(err.c_str());
            return;
            }

            String title = doc["Name"];
            Serial.println(title);

          } while (Serial.findUntil(",", "]"));
  }

}

The problem I have is that the serial monitor returns a deserialization
error on the arduino. I would appreciate your help on how to fix this, many
thanks.

bblanchon, 2020-10-13:
Which [DeserializationError](https://arduinojson.org/v6/api/misc/deserializationerror/) is it?

closed 2020-10-24 by bblanchon