bblanchon/ArduinoJson #1395: inconsistent and possibly wrong documentation for the use of ArduinoJson + StreamUtils in FAQ

ax-sc, 2020-10-07:
First of all, thank you for publishing this library and the extensive documentation of it. However, i think i found some wrong code snippets in the documentation.

In https://arduinojson.org/v6/faq/why-is-arduinojson-slow/ the use of StreamUtils for writing is done with:

```
WriteBufferingStream bufferedWifiClient(wifiClient, 64);
serializeJson(doc, bufferedWifiClient);
bufferedWifiClient.flush();
```
and for reading with:
```
ReadBufferingStream bufferedFile(file, 64);
deserializeJson(doc, bufferedFile);
```

**These code snippets are compilabe but when using them to send serialized ArduinoJson from one device over I2C and then deserialize it with another one, the last element of the Json is dropped.** I wasn´t able to fix this behaviour until i found what i suspect to be an error in the documentation.

The documentation of StreamUtils (https://github.com/bblanchon/ArduinoStreamUtils) shows examples for the functions used before, here curly brackets are used instead of normal ones when constructin the buffered element:
```
File file = SPIFFS.open("example.json", "r");
ReadBufferingStream bufferedFile{file, 64};  
deserializeJson(doc, bufferedFile);
```
and for reading:
```
WriteBufferingStream bufferedWifiClient{wifiClient, 64};
serializeJson(doc, bufferedWifiClient);
bufferedWifiClient.flush(); 
```

When adapting this snippets to my usecase, the transfer of the serialized Json over I2C worked well. The difference and therefore the changes i am proposing to the documentation are:

```diff
- WriteBufferingStream bufferedWifiClient(wifiClient, 64);
+ WriteBufferingStream bufferedWifiClient{wifiClient, 64};
serializeJson(doc, bufferedWifiClient);
bufferedWifiClient.flush();
```
and 
```diff
- ReadBufferingStream bufferedFile(file, 64);
+ ReadBufferingStream bufferedFile{file, 64};
deserializeJson(doc, bufferedFile);
```

bblanchon, 2020-10-08:
Hi @ax-sc,

These are two alternative syntaxes for calling the construction.
In this case, they are equivalent, except that the one with the braces is only available since C++11.

I don't see how this could influence the runtime, and I'm sure the generated executable is the same.
There must be another reason why it worked in one case and not the other.

Best regards,
Benoit

ax-sc, 2020-10-08:
Hi @bblanchon,

thank you for clarifying this. I must have overlooked some other changes leading to the fix of the behaviour in my case. 

Nevertheless, using the constructor consistently in the example snippets may be a good idea to avoid any confusion about this?

bblanchon, 2020-10-08:
I used the parent in arduinojson.org to be consistent with the rest of the documentation.
Indeed, on arduinojson.org, I use the old syntax because ArduinoJson is compatible with C++03.

StreamUtils, on the other hand, requires C++11, that's why I used the modern syntax.
I could change the documentation there, but I'm afraid it would look like a function call and confuse new users even more.

ax-sc, 2020-10-08:
Okay, i didn´t have a look on the required C++ versions before, but as you described the dependencies, the actual way you are dealing with this in the documentation seems to be the better solution to me than the one proposed by me in this issue.

closed 2020-10-08 by ax-sc