bblanchon/ArduinoJson #1398: Question: Problems on extracting/deserializing JSON from HTTP Header

rsenina, 2020-10-09:
Hi!

 I'm doing a HTTP Request with a Web API that returns a JSON. I could successfully get the returned HTTP Headers, but having problem on extracting the JSON result. 

Here's the look of my HTTP Header as I'm trying to print it on a serial monitor:

09:21:04.671 -> HTTP/1.1 200 OK
09:21:04.671 -> Cache-Control: no-cache
09:21:04.704 -> Pragma: no-cache
09:21:04.704 -> Content-Type: application/json; charset=utf-8
09:21:04.774 -> Expires: -1
09:21:04.774 -> Server: Microsoft-IIS/10.0
09:21:04.808 -> X-AspNet-Version: 4.0.30319
09:21:04.842 -> X-Powered-By: ASP.NET
09:21:04.876 -> Date: Fri, 09 Oct 2020 01:21:04 GMT
09:21:04.910 -> Connection: close
09:21:04.910 -> Content-Length: 245
09:21:04.943 -> 
09:21:04.943 -> [
09:21:04.943 ->   {
09:21:04.943 ->     "deviceId": "4eb22ce4-5399-43c6-a37c-4d3ea2b63b08",
09:21:05.012 ->     "deviceInfo": "DEV-092021-004",
09:21:05.046 ->     "studyID": "71afe68b-8d14-480a-ab5d-d08588577949",
09:21:05.114 ->     "tankId": "1280ff8d-e4b8-4eac-9646-ba6d032e60ab",
09:21:05.182 ->     "readingInt": 1200.0
09:21:05.182 ->   }
09:21:05.217 -> ]

I'm getting an "Unexpected response" for checking the HTTP status if I'm going to print the HTTP Header on Serial Monitor. 

```
  // Check HTTP status
  char status[32] = {0};
  client.readBytesUntil('\r', status, sizeof(status));
  if (strcmp(status, "HTTP/1.1 200 OK") != 0) {
    Serial.print(F("Unexpected response: "));
    Serial.println(status);
    return;
  }
```

For skipping the HTTP headers here is the code
```
  // Check HTTP status
  char status[32] = {0};
  client.readBytesUntil('\r', status, sizeof(status));
  if (strcmp(status, "HTTP/1.1 200 OK") != 0) {
    Serial.print(F("Unexpected response: "));
    Serial.println(status);
    return;
  }
```
Removing the code for printing the HTTP Header pass through all the conditions but extracting the values doesn't return any values at all.

for printing HTTP Header
```
    while(client.connected()) {
      if(client.available()){
        // read an incoming byte from the server and print it to serial monitor:
        char c = client.read();
        Serial.print(c);
      }
    }
```

```
 // Allocate the JSON document
 ### // Use arduinojson.org/v6/assistant to compute the capacity.
//Array size based on the assistant after i paste the returned JSON in HTTP Header.
  const size_t capacity = JSON_OBJECT_SIZE(5) + JSON_ARRAY_SIZE(5) + 200;
  DynamicJsonDocument doc(capacity);

  // Parse JSON object
  DeserializationError error = deserializeJson(doc, client);
  if (error) {
    Serial.print(F("deserializeJson() failed: "));
    Serial.println(error.c_str());
    return;
  }

// Extract values
  Serial.println(F("Response:"));
  Serial.println(doc["deviceId"].as<char*>());
  Serial.println(doc["deviceInfo"].as<char*>());
  Serial.println(doc["studyId"].as<char*>());
  Serial.println(doc["tankId"].as<char*>());
  Serial.println(doc["readingInt"].as<int>());

  // Disconnect
  client.stop();
```

I don't know if it matches the ending of my HTTP Header.

Hope you could help me with this one.

I'm using ArduinoJson v6.16.1 and running it on Arduino Mega with a connected Ethernet Shield.

bblanchon, 2020-10-09:
Hi,

Indeed, when you print the headers to Serial, you consume the HTTP stream, so the code cannot find the end of the headers.

What you need to do it let the program run and print the content of the stream at the same time.
Please check out [How to debug data coming from a stream?](https://arduinojson.org/v6/how-to/debug-data-coming-from-a-stream/) for a solution.

Best regards,
Benoit

rsenina, 2020-10-09:
Thank you for this one! I was able to resolve my issue.

closed 2020-10-09 by rsenina