bblanchon/ArduinoJson #1399: Append to a file example

tkerby, 2020-10-11:
I'm using Arduino to gather sensor data and save to SD card including GPS data and the values from various sensors. I want to store an array type document with multiple pairs of key and value e.g latitude=x, longitude=y, sensor1=z, time=hhmmss which will then repeat for each reading

Due to the volume of data I plan to save to an SD card with a new file opened each hour of operation. Unfortunately there isn't enough memory to store a complete document and append readings in the ESP32 so I'd like to dump to file every few seconds. 

I know I can serialize out to a file but how can I arrange this to append readings to the existing data? Easy enough to do with a csv file but not sure how to go about this with json. 

Any example code for this?

Thanks

bblanchon, 2020-10-12:
Hi Tim,

I recommend that you use [line-delimited JSON](https://en.wikipedia.org/wiki/JSON_streaming#Line-delimited_JSON): when you need to add a record, open the file in _append_ mode, call [serializeJson()](https://arduinojson.org/v6/api/json/serializejson/), and add a line break.
This technique is an implementation of "JSON Streaming"; you read more on that in the "Advanced Techniques" chapter of [Mastering ArduinoJson](https://arduinojson.org/book/).

Best regards,
Benoit

tkerby, 2020-10-12:
Thanks for the pointers - I've gone for line JSON on this one. Is it better to serialize to a string first or write straight to a file (possibly using your stream buffering)?

bblanchon, 2020-10-13:
I usually recommend parsing directly from the stream, but as always, your mileage may vary.
If you're interested in a particular metric (code size, speed, RAM usage...), you should better measure and compare.

closed 2020-10-24 by bblanchon