bblanchon/ArduinoJson #1400: Issue when Element contains Hyphen

1CM69, 2020-10-11:
I am successfully using ArduinoJSON on a few of my projects & have no issues when the JSON file has date elements separated by forward slashes i.e. yyyy/mm/dd

However, I am attempting to parse this URL: https://disease.sh/v3/covid-19/gov/uk

ArduinoJson Assistant correctly shows the grouped data sets and the first (at time of writing), as:

JsonObject root_2020_10_09 = doc["2020-10-09"]; 

Now when I attempt to retrieve elements further down the tree in the above Object, for example, I get nothing.

DeathsToday = root_2020_10_09["todayDeaths"];

should store then value 87 in the INT variable DeathsToday but it actual prints a zero to the serial monitor, I am guessing it's actually NULL

I am pretty sure that it has to do the hyphens used to format the date.

I use ArduinoJSON to parse another URL which is a near  identical JSON file but the dates are formatted with forward slashes in the elements, this produces no problems at all & I can retrieve all the data I require.

Is there something very simple that I am missing?

I am using ArduinoJSON v6

Thanks

bblanchon, 2020-10-11:
Hi @1CM69,

Congratulations! You managed to scare me big time :-)

However, as far as I can tell, there is no problem with dashes.
Here is an online demo as a proof: https://wandbox.org/permlink/zCjnvS91G3RdYOQI

Did you check the result of [deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/)?

Best regards,
Benoit

1CM69, 2020-10-11:
> Hi @1CM69,
>
>
>
> Congratulations! You managed to scare me big time :-)
>
>
>
> However, as far as I can tell, there is no problem with dashes.
>
> Here is an online demo as a proof: [...]

Ha! Sorry for the scare 😬🙂

OK, no idea what issue I'm seeing then. 
I'll try out a few other things and report back. 

Regards..,

1CM69, 2020-10-11:
Well, I am truly stuck.

I just cannot seem to be able to parse this particular JSON URL.

I have others working fine.

Here's my test code if anyone can help point out my error:

```
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <ArduinoJson.h>
#include <WiFiClientSecureBearSSL.h>

// WiFi Parameters
const char* ssid = "***";
const char* password = "***";

int D7deaths, ONSdeathsNEW;
long D7deathsTotal;

void setup() {
  Serial.begin(115200);
  WiFi.begin(ssid, password);
  Serial.println();
  Serial.print("Connecting.");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);

    Serial.print(".");
  }
  Serial.println();
  Serial.print("Connected to: ");
  Serial.print(ssid);
  Serial.println();
}

void loop() {

  Serial.println(D7deaths);
  Serial.println(ONSdeathsNEW);
  Serial.println(D7deathsTotal);

  
  getData();

  Serial.println(D7deaths);
  Serial.println(ONSdeathsNEW);
  Serial.println(D7deathsTotal);
  
  delay(3600000);
}

void getData() {

  std::unique_ptr<BearSSL::WiFiClientSecure>client(new BearSSL::WiFiClientSecure);

  client->setInsecure();

  HTTPClient https1;
  Serial.print("[HTTPS] begin...\n");
  if (https1.begin(*client, "https://disease.sh/v3/covid-19/gov/uk"))
  {
    Serial.print("[HTTPS] GET...\n");

    int httpCode1 = https1.GET();

    // httpCode will be negative on error
    if (httpCode1 > 0) {
      // HTTP header has been send and Server response header has been handled
      Serial.printf("[HTTPS] GET... code: %d\n", httpCode1);

      // file found at server
      if (httpCode1 == HTTP_CODE_OK || httpCode1 == HTTP_CODE_MOVED_PERMANENTLY || httpCode1 == HTTP_CODE_PERMANENT_REDIRECT) {

        const size_t capacity1 = 221*JSON_OBJECT_SIZE(13) + JSON_OBJECT_SIZE(221) + 2880;

        DynamicJsonDocument doc1(capacity1);
        deserializeJson(doc1, https1.getString());

        Serial.println(https1.getString());
        JsonObject deaths7daysago = doc1["2020-10-04"];
        D7deaths = deaths7daysago["todayDeaths"]; // on this day
        D7deathsTotal = deaths7daysago["totalDeaths"]; // cumulative total to date
        
        JsonObject ONSdeaths = doc1["2020-09-25"];
        ONSdeathsNEW = ONSdeaths["ONSweeklyDeaths"];
      }
    } else {
      Serial.printf("[HTTPS] GET... failed, error: %s\n", https1.errorToString(httpCode1).c_str());
    }

    https1.end();
  } else {
    Serial.printf("[HTTPS] Unable to connect\n");
  }

  //***
}
```

The variables should be returning:

D7deaths: 33
D7deathsTotal: 42350
ONSdeathsNEW: 234

regards..,

bblanchon, 2020-10-12:
Hi @1CM69,

I bet that you're running out of RAM.
Please check the result of [deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/).

See also:

* [How to deserialize a very large document?](https://arduinojson.org/v6/how-to/deserialize-a-very-large-document/)
* [How to use ArduinoJson with HTTPClient?](https://arduinojson.org/v6/how-to/use-arduinojson-with-httpclient/)

Best regards,
Benoit

1CM69, 2020-10-12:
> Hi @1CM69,
>
> I bet that you're running out of RAM.
> Please check the result of [deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/).
>
> See also:
>
> * [How to deserialize a very large [...]

OK, after much hassle I did find as you suggested that my ESP8266 was indeed OOM when attempting to parse this long JSON file.

I went back to basics & the actual source of the data & found that instead of getting a complete list of data I can specify a date as a filter.
The URL is very long but that is what is required:

[LINK](https://api.coronavirus.data.gov.uk/v1/data?filters=areaType=overview;date=2020-09-25&structure=%7B%22areaType%22:%22areaType%22,%22areaName%22:%22areaName%22,%22areaCode%22:%22areaCode%22,%22date%22:%22date%22,%22newDeaths28DaysByPublishDate%22:%22newDeaths28DaysByPublishDate%22,%22cumDeaths28DaysByPublishDate%22:%22cumDeaths28DaysByPublishDate%22,%22newOnsDeathsByRegistrationDate%22:%22newOnsDeathsByRegistrationDate%22,%22cumOnsDeathsByRegistrationDate%22:%22cumOnsDeathsByRegistrationDate%22%7D&format=json)

Much easier to work with this shorter JSON file in Arduino Assistant.

After changing my sketch to reflect this change I am experiencing further errors with deserialization, namely INVALID INPUT.

I read up on InvalidInput error & figured that I needed to use the http.useHTTP10(true), which I did & then uploaded the sketch again.

This time with http.useHTTP10(true) I get another error, this time: INCOMPLETE INPUT.

Here's my updated sketch, any pointers would be great.

```
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <ArduinoJson.h>
#include <WiFiClientSecureBearSSL.h>

// WiFi Parameters
const char* ssid = "The-Evans's";
const char* password = "ZZ@dbe53b10pcsXX*";

int D7deaths, ONSdeathsNEW;
long D7deathsTotal;

void setup() {
  
  Serial.begin(115200);
  WiFi.begin(ssid, password);
  Serial.println();
  Serial.print("Connecting.");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);

    Serial.print(".");
  }
  Serial.println();
  Serial.print("Connected to: ");
  Serial.print(ssid);
  Serial.println();
}

void loop() {

  Serial.println(D7deaths);
  Serial.println(ONSdeathsNEW);
  Serial.println(D7deathsTotal);

  getData();

  Serial.println(D7deaths);
  Serial.println(ONSdeathsNEW);
  Serial.println(D7deathsTotal);

  delay(3600000);
}

void getData() {

  std::unique_ptr<BearSSL::WiFiClientSecure>client(new BearSSL::WiFiClientSecure);

  client->setInsecure();

  HTTPClient https1;
  //https1.useHTTP10(true); // <-- TRYING THIS GIVE [INCOMPLETE INPUT ERROR]
  Serial.print("[HTTPS] begin...\n");
  if (https1.begin(*client, "https://api.coronavirus.data.gov.uk/v1/data?filters=areaType=overview;date=2020-09-25&structure=%7B\"areaType\":\"areaType\",\"areaName\":\"areaName\",\"areaCode\":\"areaCode\",\"date\":\"date\",\"newDeaths28DaysByPublishDate\":\"newDeaths28DaysByPublishDate\",\"cumDeaths28DaysByPublishDate\":\"cumDeaths28DaysByPublishDate\",\"newOnsDeathsByRegistrationDate\":\"newOnsDeathsByRegistrationDate\",\"cumOnsDeathsByRegistrationDate\":\"cumOnsDeathsByRegistrationDate\"%7D&format=json"))
  {
    Serial.print("[HTTPS] GET...\n");

    int httpCode1 = https1.GET();

    // httpCode will be negative on error
    if (httpCode1 > 0) {
      // HTTP header has been send and Server response header has been handled
      Serial.printf("[HTTPS] GET... code: %d\n", httpCode1);

      // file found at server
      if (httpCode1 == HTTP_CODE_OK || httpCode1 == HTTP_CODE_MOVED_PERMANENTLY || httpCode1 == HTTP_CODE_PERMANENT_REDIRECT) {

        const size_t capacity1 = JSON_ARRAY_SIZE(1) + JSON_OBJECT_SIZE(4) + JSON_OBJECT_SIZE(5) + JSON_OBJECT_SIZE(8) + 1920;

        DynamicJsonDocument doc1(capacity1);
        DeserializationError err = deserializeJson(doc1, https1.getString());

        if (err) {
          Serial.print(F("deserializeJson() failed with code "));
          Serial.println(err.c_str());
          }

        Serial.println();

        Serial.println(https1.getString());
        
        JsonObject deaths7daysago = doc1["data"][0];
        D7deaths = deaths7daysago["newDeaths28DaysByPublishDate"]; // on this day
        D7deathsTotal = deaths7daysago["cumDeaths28DaysByPublishDate"]; // cumulative total to date
        
        JsonObject ONSdeaths = doc1["data"][0];
        ONSdeathsNEW = ONSdeaths["newOnsDeathsByRegistrationDate"];
      }
    } else {
      Serial.printf("[HTTPS] GET... failed, error: %s\n", https1.errorToString(httpCode1).c_str());
    }

    https1.end();
  } else {
    Serial.printf("[HTTPS] Unable to connect\n");
  }

  //***
}
```
thanks

bblanchon, 2020-10-13:
Hi @1CM69,

[IncompleteInput](https://arduinojson.org/v6/api/misc/deserializationerror/#incompleteinput) usually means "time-out".
Try to [increase the stream's timeout](https://arduinojson.org/v6/how-to/change-the-timeout/).

BTW, the response is much smaller now, so you may consider going back to your original implementation using `getString()`.

Best regards,
Benoit

closed 2020-10-24 by bblanchon