bblanchon/ArduinoJson #1401: Deserialize an array with hexadecimal values

joelzavala, 2020-10-11:
board: Esp8266 NodeMCU v1

hello everyone, i have a small inconvenience in trying to deserialize some json values.

I tell you, in my esp8266 made a post with the HTTPClient library and I get (as a response from my server) a string like this:
{"a": [0x5b, 0x70, 0x88], "b": 0x45, 0x54, 0x55]}

When I try to deserialize it I get this error deserializeJson() failed: InvalidInput. I would like to know how to deserialize that array to put it in a uint8_t variable

Thanks in advance :)

bblanchon, 2020-10-12:
Hi @joelzavala,

Indeed, JSON doesn't allow hexadecimal values.
JSON5 allows hexadecimal values, but ArduinoJson doesn't support this format.

Best regards,
Benoit

joelzavala, 2020-10-12:
Hello again, now I have a different question, how do I copy an array from doc["on"] to uint8_t?

Thanks in advance :)

bblanchon, 2020-10-13:
See [copyArray()](https://arduinojson.org/v6/api/misc/copyarray/)

closed 2020-10-24 by bblanchon