bblanchon/ArduinoJson #1403: PubSubClient Payload Serialization exit OK but could not access data.

CristoJV, 2020-10-12:
I'm receiving this json mqtt payload.
"{\"red\":100,\"g\":100,\"b\":100,\"wc\":100,\"ww\":100}"
I deserialize the (bytes *) payload using the provided library method deserializeJSON, that return no errors.
Then after I tried to access the "red" key-value, but I'm only getting -1.

This is my handle method:
```
void handle_msg(char *topic, byte *payload, unsigned int length) {
      StaticJsonDocument<256> doc;
      DeserializationError err = deserializeJson(doc, payload, length);
      Serial.println(err.c_str());
      int r = object["red"] | -1;
      Serial.println(r);
      Serial.print(object["red"].is<int8_t>());
      Serial.print(object["red"].is<int16_t>());
      Serial.print(object["red"].is<int32_t>());
      Serial.print(object["red"].is<char*>());
      Serial.print(object["red"].is<double>());
      Serial.print(object["red"].is<float>());
}

```
Returns
Ok
-1
000000

CristoJV, 2020-10-12:
I could find a trick solution.
First I create a String with the payload.
```
String str = String((char*)payload);
```
then I converted to a char array.
```
char output[100];
str.toCharArray(output, length);
```
then I deserialize the JSON
```
deserializeJson(doc, output, length);
int r = doc["red"] | -1;
Serial.println(r);
```
And It works.

SERIAL PRINTS____

```
100
```

bblanchon, 2020-10-13:
Hi @CristoJV,

The following JSON document is a string, not an object:

```json
"{\"red\":100,\"g\":100,\"b\":100,\"wc\":100,\"ww\":100}"
```

The object looks like this:

```json
{"red":100,"g":100,"b":100,"wc":100,"ww":100}
```

You can extract the object from the original string with ArduinoJson, but you need to call [deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/) twice:

```c++
StaticJsonDocument<100> doc1;
deserializeJson(doc1, json);

StaticJsonDocument<200> doc2;
deserializeJson(doc2, doc1.as<const char*>());

int r = doc["red"] | -1;
```

Best regards,
Benoit

bblanchon, 2020-10-13:
PS: [demo on wandbox.org](https://wandbox.org/permlink/AWHlKiSFmMalMZYU)

CristoJV, 2020-10-13:
Thank you Benoit, but why do I have to call it twice?
Does the deserialization perform different tasks when it is called with a String or with a JSON object?
Is it justified the use of more memory just to be able to deserialize the JSON String?

Thank you

bblanchon, 2020-10-14:
Usage is unusual because the input is unusual; you would have to do the same with any library.
To simplify the program and reduce memory consumption, you should repair the input.

closed 2020-10-24 by bblanchon