bblanchon/ArduinoJson #1407: Question:  Stream JSON as Response to WebServerRequest

lbussy, 2020-10-13:
In this case, I am using ESPAsyncWebServer, but I believe the syntax is the same for ESPWebServer.

I am looking to serialize the JSON directly to the connection without taking the intermediate step of serializing to a String.  Is this possible?  I've Googled for a bit and not hit the right combination of terms to find a solution.

This is how I am doing it now:

``` cpp
    server.on("/thisVersion/", HTTP_ANY, [](AsyncWebServerRequest *request) {
        const size_t capacity = JSON_OBJECT_SIZE(1);
        StaticJsonDocument <capacity> doc;

        doc["version"] = version();

        String json;
        serializeJsonPretty(doc, json);
        request->header("Cache-Control: no-store");
        request->send(200, F("application/json"), json);
    });
```

bblanchon, 2020-10-14:
Hi Lee,

I don't know of any better way.

On the one hand, ArduinoJson can *write* to a stream but doesn't provide a stream that can be read from.
On the other hand, ESPAsyncWebServer can *read* from a stream but doesn't provide a stream that can be written to.
I believe that an intermediate stream with an internal buffer could solve the impedance mismatch, but it would have to regularly pause the writing process (most likely with setjmp/longjmp).
I never took the time to push this idea further because I'm afraid it would be very brittle.

Best regards,
Benoit

lbussy, 2020-10-14:
Well, I guess I don’t feel too bad that I did not figure it out myself this time.  Thank you for the answer!

closed 2020-10-14 by lbussy