bblanchon/ArduinoJson #1410: Filtering from JSON Array

jomiva, 2020-10-21:
Hey, I've seen others issues about filtering and JsonFilterExample.ino but I couldn't solve my problem.
I have a Json Array (keys.json) which has an array of objects. Example:
[
  {
    "code": 268435455,
    "name": "Maria",
    "status": false,
  },
  {
    "code": 1234,
    "name": "Jose",
    "status": true,
  }
]

If I try to deserialize it without filtering, it works fine. But if I try to filter "code" and "status", returns null.
This is my filter.
StaticJsonDocument<64> filter;
filter["codigo"] = true;
filter["status"] = true;
deserializeJson(doc, llavesFile, DeserializationOption::Filter(filter));

The difference between this and other issues is that I'm filtering from an Json Array and not from a Json Document which has an array inside. 

My code
[image]

bblanchon, 2020-10-22:
Hi,

The input document is an array, so the filter must be an array too:

```c++
filter[0]["codigo"] = true;
filter[0]["status"] = true;
```

See:

* [`deserializeJson()`, Filtering](https://arduinojson.org/v6/api/json/deserializejson/#filtering)
* [JsonFilterExample.ino](https://arduinojson.org/v6/example/filter/)
* [ArduinoJson 6.15: Filtering done right](https://arduinojson.org/news/2020/03/22/version-6-15-0/)

Best regards,
Benoit

jomiva, 2020-10-23:
Thanks! I thought I needed to filter it with an object.

closed 2020-10-23 by jomiva

bblanchon, 2020-10-24:
You're welcome @jomiva. Good luck with your project!
Don't hesitate to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project 😉.