bblanchon/ArduinoJson #1411: Problem with 6.17.0 with default value using overloaded "|" operator

labodj, 2020-10-21:
Hi,
since 6.17.0 this code is not working, it does not compile because it's complaining about the ambiguity of | overloaded operator.

```c++
#include <Arduino.h>
#include <ArduinoJson.h>

void setup()
{
}

void loop()
{
  const size_t capacity = JSON_OBJECT_SIZE(1) + 20;
  DynamicJsonDocument doc(capacity);
  const char *json = "{\"sensor\":\"gps\"}";
  deserializeJson(doc, json);
  const char *test = "test";
  const char *sensor = doc["sensor"] | test; // "gps"
}
```

Commenting lines 30-37 in /src/ArduinoJson/Variant/VariantOperators.hpp solve this problem, but can cause other unexpected ones.

closed 2020-10-22 by bblanchon

bblanchon, 2020-10-22:
Hi @labodj,

Thank you very much for reporting this bug.
The fix is available on the "6.x" branch.
I'll publish it in the next revision in a few days.

Best regards,
Benoit

bblanchon, 2020-11-07:
The fix is published in [ArduinoJson 6.17.1](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.17.1)

commit 2664a2d0 (2020-10-22): Fixed error "ambiguous overload for operator|" (fixes #1411)