bblanchon/ArduinoJson #1412: [proposal] add is<JsonArray/Object/VariantConst>()

jjsuwa-sys3175, 2020-10-22:
currently, to get a read-only JSON aggregate from its parent variant/document is such as:
```
JsonVariant var=doc[0];
if (var.is<JsonArray>()) {
  auto ro_arr=var.as<JsonArrayConst>();
  ...
}
```
in terms of the functional orthogonality, that can be:
```
JsonVariant var=doc[0];
if (var.is<JsonArrayConst>()) {
  auto ro_arr=var.as<JsonArrayConst>();
  ...
}
```

bblanchon, 2020-10-24:
Hi @jjsuwa-sys3175,

You're absolutely right.
Unfortunately, this is a breaking change, so I'll have to wait a little before fixing this.

Example:

```c++
JsonVariantConst var = doc[0];
bool isArray = var.is<JsonArray>();
```

With the current implementation, `isArray` can be `true`, but after the change, it will always be `false`, because we cannot get a `JsonArray` from a `JsonVariantConst`.
We have the same problem with a `const JsonDocument&`.

Best regards,
Benoit

jjsuwa-sys3175, 2020-10-24:
@bblanchon :
> With the current implementation, isArray can be true, but after the change, it will always be false,

i guess it should emit compile-time error rather than return `false` because of early-stage error awareness development, and you also said:
> we cannot get a `JsonArray` from a `JsonVariantConst`.

closed 2021-02-26 by bblanchon

bblanchon, 2021-05-05:
Fix published in [ArduinoJson 6.18.0](https://arduinojson.org/news/2021/05/04/version-6-18-0/)

commit 3aab36fe (2021-02-25): Changed JsonVariantConst::is<JsonArray/JsonObject>() to return false (closes #1412)