bblanchon/ArduinoJson #1414: bug: segmentation fault: adding a few members after a big array (~4000) member

ah-terr, 2020-10-22:
amd64, gcc version 7.5.0 (Ubuntu 7.5.0-3ubuntu1~18.04)

```
#include <iostream>
#include <string>

#include "ArduinoJson.h"

using namespace ArduinoJson;

struct arrElement
{
    uint8_t type;
    uint32_t number;
    int64_t time;
    uint16_t measures;
    uint32_t max;
    uint32_t min;
    std::string descr;
    bool ok;
};

int main()
{
    const int max = 4100;
    const auto memSize = JSON_ARRAY_SIZE(max) + JSON_OBJECT_SIZE(1) + JSON_OBJECT_SIZE(6) + max * JSON_OBJECT_SIZE(7);
    DynamicJsonDocument doc(memSize * 2);
    auto data = doc.to<JsonObject>().getOrAddMember("data");
    auto array = data.getOrAddMember("array").to<JsonArray>();
    for (int i = 0; i < max; ++i)
    {
        arrElement el{1, 5, 1603393921, 11, 19, 1, "description", true};

        auto nested = array.createNestedObject();
        nested["type"] = el.type;
        nested["num"] = el.number;
        nested["time"] = el.time;
        nested["measures"] = el.measures;
        nested["min"] = el.min;
        nested["max"] = el.max;
        nested["ok"] = el.ok;
    }
    
    data.getOrAddMember("quantity").set(max);
    data.getOrAddMember("day").set(15u);
    data.getOrAddMember("people").set(11);
    data.getOrAddMember("tools").set(7);
    data.getOrAddMember("time").set(1603394921);

    std::string che;
    serializeJson(doc, che);
    
    std::cout << che << '\n';

    return 0;
}
```
output:
```
Segmentation fault (core dumped)
```
Producing array after an other members makes the program good.

bblanchon, 2020-10-23:
Hi @ah-terr,

Thank you very much for reporting this but.

As you probably know, ArduinoJson uses a linked list to store the nodes of the JSON document.
To reduce the memory footprint, instead of saving the *address* of the next node, each node saves the *offset* of the next node.
In this case, instead of using a 64-bit pointer, it uses a 16-bit integer, saving 6 bytes per node.
Unfortunately, this means that two nodes cannot be more than 32767 slots apart.
This is a safe assumption on an embedded platform, but in your case, the members `"array"` and `"quantity"` are more distant than that.

[I added an assert](https://github.com/bblanchon/ArduinoJson/commit/6738b71ef8e9237a4d2b09bf26759333e5ec8f9a) to detect the problem more easily, but I'm not sure what else I can do.
Sure, I could add a rule to use an `int32_t` offset on 64-bit platforms, but does it make any sense?
It would solve your particular issue, but it could hide a problem when compiled for the target platform.

What do you think?
Your `JsonDocument` is 2MB large; do you really plan to run this code on an embedded platform?
Why did you write this test in the first place?

Best regards,
Benoit

ah-terr, 2020-10-23:
Hi @bblanchon  , glad to read your fast answer!
Concerning the 2MB json: guess, this value will gradually be gone to past. And in addition as far as I know, despite of ArduinoJson embedded purpose, it successfully uses on other platforms.

Imho, using of int32_t could be brought into explicit config directives.

bblanchon, 2020-10-24:
Hi @ah-terr,

It's not just 2MB. It's more than 32767 nested values in a single array!
Even if you manage to fit this in an embedded platform, the performance would be terrible.
As I said, ArduinoJson uses linked lists, so accessing an element is done in O(n).

You're right; eventually, 2MB will seem small, but right now, 8-bits microcontrollers are still a thing (2KB of RAM for a UNO).
Squeezing every possible byte of RAM of those CPUs is one of the missions of ArduinoJson 6.
Obviously, the strategy will have to change in the future.

"Why did you write this test in the first place?"
In other words, is this a real use case or just a benchmark?
If that's a real use case, then sure, I can add a configuration directive.

Best regards,
Benoit

ah-terr, 2020-10-24:
Hi @bblanchon ,
sure, I understand we pay by performance for the good capacity.
It is a real use case: I have a two parts of project linked between each other through your lib. One of them is on the STM. Other one -- a host  -- not yet a desktop pc, but let's say something more powerfull than Raspberry. And this long array is accumulated on the host part and then is sent to internet.

Best regards

bblanchon, 2020-10-26:
Hi @ah-terr,

Thanks for the explanation.
The easiest solution is to use a 16-bit index only when `ARDUINOJSON_EMBEDDED_MODE` is true.
I'll try to add this in 6.18.

Best regards,
Benoit

ah-terr, 2020-10-26:
Hi @bblanchon ,

Good news, thank you.
As far as I understand, replacing `VariantSlotDiff` type would not lead to any unpredictable consequences? And the memory size estimator would work properly, right?

bblanchon, 2020-10-27:
Hi @ah-terr,

If you set `VariantSlotDiff` to `int32_t`, everything should work fine, but results from the [ArduinoJson Assistant](https://arduinojson.org/v6/assistant/) will be too low.
The macros `JSON_ARRAY_SIZE()` and `JSON_OBJECT_SIZE()` will continue to work, though.

Best regards,
Benoit

closed 2020-10-29 by bblanchon

bblanchon, 2020-11-07:
The fix was published in [ArduinoJson 6.17.1](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.17.1)

commit c711fe59 (2020-10-28): Allowed more than 32767 values in non-embedded mode (fixes #1414)