bblanchon/ArduinoJson #1415: Another error after #1411 fixed

labodj, 2020-10-22:
Hi @bblanchon, after you fixed the code for the issue #1411 I posted yesterday I've discovered another problem (keep in mind that I'm testing with upstream code @ ff66182dc648dedd4959239232ebd2f90039066b), I'm using this library in conjunction with [homie-esp8266](https://github.com/homieiot/homie-esp8266), more precisely with [my fork](https://github.com/labodj/homie-esp8266/tree/feature-option-retain-set-topic)

They [committed](https://github.com/homieiot/homie-esp8266/commit/90b887248185f11df523e27905797cff71463d83) yesterday to stay on an older ArduinoJson version because 6.17.0 has some problems with their library, I think it's not so smart not opening an issue here, (it's one of the reason why I am mantaining a fork), one was #1411, the other one is this:

```
.pio/libdeps/release/Homie/src/Homie/Utils/Validation.cpp:374:84:   required from here
.pio/libdeps/release/ArduinoJson/src/ArduinoJson/Variant/VariantAs.hpp:64:55: error: no type named 'type' in 'struct ArduinoJson6170_91::enable_if<false, ArduinoJson6170_91::ObjectRef>'
```

more precisely I can trigger the error like so:

```c++
#include <Arduino.h>
#include <ArduinoJson.h>

void setup()
{
}

void loop()
{
  StaticJsonDocument<JSON_OBJECT_SIZE(1)> doc;

  JsonObject object = doc.to<JsonObject>();
  object["hello"] = "world";

  StaticJsonDocument<0> emptyDoc;
  JsonObject anotherObject = object["hello"] | emptyDoc.to<JsonObject>();
}
```

bblanchon, 2020-10-23:
Hi @labodj,

This should be fixed on branch "6.x".
Thanks for reporting this issue.

Best regards,
Benoit

closed 2020-10-23 by bblanchon

bblanchon, 2020-11-07:
The fix was published in [ArduinoJson 6.17.1](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.17.1)

commit 10ec0f21 (2020-10-23): Fixed `operator|(MemberProxy, JsonObject)` (fixes #1415)